garywill/linux-router #1: Run as service?

gitxmax, 2018-12-28:
Hello, thank you for your job.
I tried running this (on Raspberry 3, with 2 external usb dongles), as well as the original create_ap and just like the original script, it seems to have an issue, with the interfaces, if run from crontab.
Any chance to run it as a service, so it will be much easier, to run it at start up?
Also both the script seem to have an issue, with wlan0 -> wlan1 relay. The speed is cut drastically, best I can get is 50% less download rate on wlan1, compared to wlan0. The upload speed seems unaffected that much.
I know the hardware can perform a lot better, since running the LEDE, on the set up, only has around 10% downgrade.
For additional information, on-board wifi disabled, as well as network-manager is off. Clean install of debian.

Thanks again.

diveyez, 2019-05-05:
This is needed right now!
 Thank you so much!

eschwelgin, 2021-03-07:
This is what I'm working on, but I can't get the router to launch in a usable state automatically on boot using any of the three methods that follow

It will launch and broadcast an ap, but nothing will connect to it. 

If I run this from cli it works as expected
$ sudo ./lnxrouter --random-mac --ap wlan0 netgear0 -p testpass -c 3

If I try to set it as a systemd service things look good but nothing can connect to the ap (worth noting that starting the service manually after boot produces no change, enabling, disabling, stopping / starting, etc.. ) 

router.service - linux-router
     Loaded: loaded (/etc/systemd/system/router.service; disabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-03-07 22:19:40 UTC; 2s ago
   Main PID: 3091 (lnxrouter)
      Tasks: 7 (limit: 4915)
     CGroup: /system.slice/router.service
             ├─3091 /bin/bash /home/ubuntu/Downloads/linux-router-master/lnxrouter --ap wlan0 netgear0
             ├─3202 /bin/bash /home/ubuntu/Downloads/linux-router-master/lnxrouter --ap wlan0 netgear0
             ├─3205 sleep 2
             ├─3206 hostapd -P /dev/shm/lnxrouter_tmp/lnxrouter.wlx984827adde1a.conf.3u7/hostapd.pid /dev/shm/lnxrouter_tmp/lnxrouter.wlx984>
             ├─3207 /bin/bash /home/ubuntu/Downloads/linux-router-master/lnxrouter --ap wlan0 netgear0
             ├─3208 sleep 3
             └─3209 sleep 10

Mar 07 22:19:40 ubuntu linux-router[3091]: Use random LAN IPv4 address 192.168.247.1
Mar 07 22:19:40 ubuntu linux-router[3091]: WARN: Wifi is not protected by password
Mar 07 22:19:40 ubuntu linux-router[3091]: Set wlan0 unmanaged by NetworkManager
Mar 07 22:19:41 ubuntu linux-router[3091]: haveged_watchdog PID: 3202
Mar 07 22:19:41 ubuntu linux-router[3091]: Starting hostapd
Mar 07 22:19:41 ubuntu linux-router[3091]: hostapd PID: 3206
Mar 07 22:19:41 ubuntu linux-router[3206]: Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.3u7/hostapd.conf
Mar 07 22:19:42 ubuntu linux-router[3206]: Using interface wlan0 with hwaddr 98:48:72:da:de:ba and ssid "netgear0"
Mar 07 22:19:42 ubuntu linux-router[3206]: wlan0: interface state UNINITIALIZED->ENABLED
Mar 07 22:19:42 ubuntu linux-router[3206]: wlan0: AP-ENABLED

I've tried a number of changes in the .service file but there has been no change. Referanced the create-ap .service file too. I can't say I've tried a conclusive set of params, but .service is where I've spent most of my time. Worth noting is I've tried these w/o auth, channel, and random-mac params. no change

[Unit]
Description=linux-router
After=network.target

[Service]
Type=simple
ExecStart=/home/ubuntu/Downloads/linux-router-master/lnxrouter --ap wlan0 netgear0
KillSignal=SIGINT
Restart=always
RestartSec=10
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=linux-router

[Install]
WantedBy=multi-user.target

I've tried rc.local, I'm sure I'm doing something wrong here because it won't launch/broadcast at all. It is  chmod +x

#!/bin/sh -e -p
#
# rc.local

/home/ubuntu/Downloads/linux-router-master/lnxrouter --random-mac --ap wlan0 netgear0 -p testpass -c 3 &

exit 0 

I've also tried starting it with 
$ gnome-session-properties

using the same command 
$ sudo ./lnxrouter --random-mac --ap wlan0 netgear0 -p testpass -c 3

Still results in a broadcast that I can't connect to. Tried several things including cell phones, computers, and an amazon fire stick (lol) 

its running on a raspi / ubuntu 20.10 server

It may be worth noting that I am using two seperate interfaces for wifi. internal is recieving, usb is broadcasting  

Any idea what I may be doing wrong here?

garywill, 2021-03-08:
@eschwelgin 
Is that all log? Didn't see iptables and dnsmasq.

> ├─3206 hostapd -P /dev/shm/lnxrouter_tmp/lnxrouter.wlx984827adde1a.conf.3u7/hostapd.pid /dev/shm/lnxrouter_tmp/lnxrouter.wlx984>
>
> Mar 07 22:19:41 ubuntu linux-router[3091]: hostapd PID: 3206
>Mar 07 22:19:41 ubuntu linux-router[3206]: Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.3u7/hostapd.conf

Why different configuration folder name? `lnxrouter.wlan0.conf.3u7` and `lnxrouter.wlx984827adde1a.conf.3u7`

You chould try adding iptables.service and networkmanager.service as dependency. Try add a 2-minutes delay before calling `lnxrouter` to to have system boot completely

> I've tried rc.local, I'm sure I'm doing something wrong here because it won't launch/broadcast at all. It is chmod +x
>
> #!/bin/sh -e -p
> /home/ubuntu/Downloads/linux-router-master/lnxrouter --random-mac --ap wlan0 netgear0 -p testpass -c 3 &
> exit 0

If use script, your script should be something like
```
sleep 120  # try a long enough delay
/home/ubuntu/Downloads/linux-router-master/lnxrouter --random-mac --ap wlan0 netgear0 -p testpass -c 3 &
sleep 2   # this sleep is must
disown -a  # must !
sleep 2  # this sleep is must
exit 0 # then you can exit
```

eschwelgin, 2021-03-08:
@garywill 

I cleaned up my op to make some of the outputs a bit clearer, and to fix a mistake. sorry for the confusion. 

> @eschwelgin
> Is that all log? Didn't see iptables and dnsmasq.

Thats the second part of the output from systemctl status, I added the commands above and deleted the space in the middle of the output to make it a bit clearer 

> Why different configuration folder name? `lnxrouter.wlan0.conf.3u7` and `lnxrouter.wlx984827adde1a.conf.3u7`

Sorry I changed all of the wlx984827adde1a to wlan0 for clarity, I must have missed that one. They are all the same 

> You chould try adding iptables.service and networkmanager.service as dependency. Try add a 2-minutes delay before calling `lnxrouter` to to have system boot completely

This produces no change. Router still broadcasts but nothing will connect to it. Here's my updated .service file 

[Unit]
Description=linux-router
After=network.target iptables.service networkmanager.service
[Service]
Type=simple
ExecStart=/home/ubuntu/Downloads/linux-router-master/lnxrouter --ap wlx984827ad>
KillSignal=SIGINT
Restart=always
RestartSec=10
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=linux-router
[Install]
WantedBy=multi-user.target

I also tried starting the service manually (disabled on startup, systemctl start) after the system had booted. Still runs with issues, broadcast ap / no connections) 

I also managed to get it running from rc.local. Surprisingly enough it was because I forgot the & at the end of the command to allow it to fork. I ran it like this for a while and it still ran with the same issues, broadcast ap / no connections. I fixed the file and booted back into grub, where I ran rc.local manually, and in that state it worked as expected. 

It really seems to me like any type of background / automatic start state results in the same issue, where it broadcasts the ap but nothing will connect to it. Any time I start it manually, through the cli or a script including rc.local) it runs as expected. 

The only other thing to note, last night I was playing around with it and although I don't remember the exact circumstances, I was seeing the errors from check_wifi_settings() "WARN: Using 'iwconfig', not as good as 'iw'" and "ERROR: Can't use 'iwconfig' to operation this WiFi interface neither". I'm sorry that I can't seem to reproduce today, but it was right before I forgot the & in rc.local and gave up for the night 

I appreciate your help

garywill, 2021-03-09:
@eschwelgin

I tried to run by systemd. Turned out `dnsmasq` failed
In dnsmasq.conf
```
log-facility=/dev/stdout
```
causes error 
```
cannot open log /dev/stdout: No such device or address
```
in systemd environment.
(You could try modifying `StandardOutput=syslog` to something else )

For a quick workaround you can delete `log-facility=/dev/stdout`

BTW, I have to fix watchdog. It didn't react when dnsmasq fails. (and it couldn't recognize zombie process)

garywill, 2021-04-17:
@eschwelgin 
Changed how dnsmasq outputs.
Now it uses named pipe. You could try it again.

lucaspeed72, 2022-10-08:
Sorry to reopen an old thread but I tried to run lnxrouter using a script:

`#!/bin/sh -e -p
sleep 120
/usr/sbin/lnxrouter -i eth0 &
sleep 2
disown -a
sleep 2
exit 0`

This script is launched by cron (@reboot) on a Raspberry Pi 4 and gives internet to an AVM Fritz Extender that should run a wifi network.
If I launch lnxrouter -i eth0 manually it works fine, but on reboot I have a running router that I cannot connect to because does not receive an IP address.
Any hint about it?
Thank you

jordanh, 2024-01-25:
> Changed how dnsmasq outputs.
> Now it uses named pipe. You could try it again.

Just because nobody else said so explicitly: this works now. Here is my service definition `lnxrouter-hotspot.service`

```
[Unit]
Description=lnxrouter Hotspot Service
After=network.target iptables.service networkmanager.service

[Service]
Type=simple
ExecStart=/usr/local/bin/lnxrouter --ap wlx40a5ef05a66a "jordan-t4pro" -p "xxxxxxxx" --no-virt
Restart=always
KillSignal=SIGINT
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=lnxrouter-hotspot
User=root
RestartSec=10

[Install]
WantedBy=multi-user.target
```

N.B. to other readers: you may not want `--no-virt` 

I came here to try this script after failing to get my USB `148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter` working in ap mode using `nmcli` with WPA-PSK (or even WEP, _shudder_).  Setting things up "the old-fashioned way" using lnxrouter worked GREAT.

Thank you for this script!