garywill/linux-router #2: 这个对NM的网络管理不管用吧？

Aquarian-Age, 2019-01-05:
试了一下 报错了
```
sudo ./lnxrouter --ap wlan0 test
PID: 10782
Target interface is wlan0
Creating a virtual WiFi interface... 
x0wlan0 created.
WARN: Wifi is not protected by password
Set x0wlan0 unmanaged by NetworkManager
haveged_watchdog PID: 10873

Starting hostapd
hostapd PID: 10877
Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.ISX/hostapd.conf
Using interface x0wlan0 with hwaddr 6c:71:d9:1b:c4:b5 and ssid "test"
x0wlan0: interface state UNINITIALIZED->ENABLED
x0wlan0: AP-ENABLED 

iptables: NAT 
MASQUERADE  all opt -- in * out !x0wlan0  192.168.18.0/24 !-> 192.168.18.0/24   /* lnxrouter-10782-x0wlan0 */
ACCEPT  all opt -- in x0wlan0 out *  192.168.18.0/24  -> 0.0.0.0/0   /* lnxrouter-10782-x0wlan0 */
ACCEPT  all opt -- in * out x0wlan0  0.0.0.0/0  -> 192.168.18.0/24   /* lnxrouter-10782-x0wlan0 */

iptables: allow DNS port access
ACCEPT  tcp opt -- in x0wlan0 out *  192.168.18.0/24  -> 192.168.18.1   tcp dpt:53 /* lnxrouter-10782-x0wlan0 */
ACCEPT  udp opt -- in x0wlan0 out *  192.168.18.0/24  -> 192.168.18.1   udp dpt:53 /* lnxrouter-10782-x0wlan0 */

iptables: allow DHCP port access
ACCEPT  udp opt -- in x0wlan0 out *  0.0.0.0/0  -> 0.0.0.0/0   udp dpt:67 /* lnxrouter-10782-x0wlan0 */
Setting /usr/sbin/dnsmasq to complain mode.

ERROR: Path doesn't start with / or variable: libvirt_leaseshelper

Starting dnsmasq
dnsmasq PID: 10938

dnsmasq: cannot read /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.ISX/dnsmasq.conf: Permission denied
Error occured

ERROR: dnsmasq exited

Error occured

Doing cleanup.. 
Killed 10873 lnxrouter
x0wlan0: interface state ENABLED->DISABLED
x0wlan0: AP-DISABLED 
nl80211: deinit ifname=x0wlan0 disabled_11b_rates=0
Killed 10877 hostapd
iptables: stop NAT
iptables: stop allowing DNS
iptables: stop dhcp
Exiting: This is the only running instance
Restore x0wlan0 managed by NetworkManager
Cleaning up done
已杀死

```

garywill, 2019-01-05:
```
Setting /usr/sbin/dnsmasq to complain mode.

ERROR: Path doesn't start with / or variable: libvirt_leaseshelper
```
这个错误发生在`complain dnsmasq`，和apparmor有关，应该和NM无关。
是什么系统？create_ap能用吗？

Aquarian-Age, 2019-01-11:
> ```
> Setting /usr/sbin/dnsmasq to complain mode.
> 
> ERROR: Path doesn't start with / or variable: libvirt_leaseshelper
> ```
> 这个错误发生在`complain dnsmasq`，和apparmor有关，应该和NM无关。
> 是什么系统？create_ap能用吗？
> 
> 搜索了一下，会不会是这个bug https://bugs.launchpad.net/apparmor/+bug/1775591

不太清楚 我openSUSE15.0系统 create_ap没用过 不清楚

closed 2019-01-27 by Aquarian-Age