garywill/linux-router #3: proxy forwarding not working

chaudhryfaisal, 2019-02-03:
thank you for your awesome effort in putting the script together. 
I am trying to test with AP and PROXY. AP works fine with following command `./lnxrouter -o eth0 --ap wlan0 AP_NAME --password AP_PASS`, I am able to connect my android device and internet is working. When I try to launch with `--tp PORT` option my phone no longer gets internet. 

I have validated my **redsocks** is setup properly by following steps 
1. `iptables -t nat -A OUTPUT  -p tcp --dport 80 -j REDIRECT --to-port 12345`
2. `curl http://yahoo.com`
in redsocks logs I see curl is making connection. 

I am using `Ubuntu Server` on VM Ware Fusion on Mac 

**lsb_release -a**
```bash
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty 
```

garywill, 2019-02-03:
what is your proxy software? what type is the proxy? this script require the proxy supports transparent proxy. have you tried redirecting all traffic to your proxy not just port 80?

chaudhryfaisal, 2019-02-03:
- using Charles Proxy
- SOCKS5 proxy type
- I dont see an option for transparent proxy under socks but I have enabled under regular proxy which runs on different port
- yes I have tried redirecting 80, 443 and even by process like `
iptables -t nat -A OUTPUT -p tcp -m owner --uid-owner root -j REDSOCKS` and it works  just not with `AP`

after many hours... I was able to get it working with **sshuttle** `sshuttle -Nr root@SERVER_2_IP 0/0 -l 0.0.0.0` but this required another server with **ssh** and ** redsocks** with following rules

```bash
# Create new chain
iptables -t nat -N REDSOCKS
# Everything should be redirected to port 12345
iptables -t nat -A REDSOCKS -p tcp -j REDIRECT --to-ports 12345
# Any tcp connection made by `root' should be redirected.
iptables -t nat -A OUTPUT -p tcp -m owner --uid-owner root -j REDSOCKS
``` 

still would love your help to get it working without sshuttle

thank you

garywill, 2019-02-03:
As I remember, iptables doesn't support redirecting to SOCKS proxy. I have tried `iptables -t nat -I OUTPUT -p tcp -j REDIRECT --to-ports xxxx` with different SOCKS proxy and failed.

> yes I have tried redirecting 80, 443 and even by process like iptables -t nat -A OUTPUT -p tcp -m owner --uid-owner root -j REDSOCKS and it works

I guess your proxy does special treatment on web requests. Did you test ports and protocols other than web? If yes, maybe it really supports transparent proxy.
check these: Is the proxy listening on right interface ?(`0.0.0.0` or `127.0.0.1`) Is firewall blocking input from AP?

Besides there's a program called `tun2socks` which  is useful if you want to combine iptables and SOCKS
> It implements a TUN device which accepts all incoming TCP connections (regardless of destination IP), and forwards the connections through a SOCKS server.

garywill, 2019-02-03:
Also found these tools useful:
https://github.com/semigodking/redsocks
https://github.com/darkk/redsocks
https://github.com/vi/tcpsocks

chaudhryfaisal, 2019-02-03:
my setup is as following 
`iptables -> port 12345 (redsocks tcp port) -> 7771 (Charles proxy SOCK5 port)` so not going from iptables to socks directly, using redsocks to bridge the gap between. 

I believe my issue is somewhere with `iptables` rules.