garywill/linux-router #4: dnsmasq: unknown user or group nobody

acmkonings, 2019-05-14:
I am experiencing above error when using 
`lnxrouter -i eth0`

on my Raspberry Pi 2B with raspbx 14 (stretch). The lnxrouter script cancels itself by consequence.
Any help would be appreciated. Thanks!

garywill, 2019-05-14:
Maybe your system doesn't have `nobody` user.

Try deleting this
https://github.com/garywill/linux-router/blob/5452234c081835715f035ca981225e7b3d5ae4f8/lnxrouter#L1427-L1428

diveyez, 2019-06-18:
No No NO. Do not delete it, instead MAKE THE GROUP AND USER. nobody nobody should have been added, but it appears that raspbx likes to do things the retarded way.

jcarius, 2019-07-24:
Same on Ubuntu 18.04. The workaround by @garywill 

> Maybe your system doesn't have `nobody` user.
> 
> Try deleting this
> 
> https://github.com/garywill/linux-router/blob/5452234c081835715f035ca981225e7b3d5ae4f8/lnxrouter#L1427-L1428

works.

garywill, 2019-07-25:
> Same on Ubuntu 18.04

That's strange. Ubuntu doesn't have `nobody`? Or is it other reason?

@jcarius Please check if your ubuntu have `nobody` user and group. It's needed for a completed Linux system.

As @diveyez said, `nobody nobody` should have been added. Workaround is just workaround.

jcarius, 2019-07-25:
A quick google search turned out that for Debian flavors one needs to use `nobody:nogroup`, see [this answer](https://askubuntu.com/a/138977)

garywill, 2019-07-25:
Just added a group judge. You can try if it works.

jcarius, 2019-07-25:
Works! I guess the issue can be closed.

closed 2024-12-30 by garywill