garywill/linux-router #6: help

l4ckyguy, 2019-08-18:
Hello! I have a problem ((I’m trying to make a router from rpi zero w, I set up the connection in this way: the Internet takes over wifi (wlan0) and gives it to the computer over ethernet-usb (usb0). everything works smoothly.

cat /etc/iptables/rules.v4

# Generated by xtables-save v1.8.2 on Sat Aug 17 20:50:44 2019
*filter
:INPUT ACCEPT [462:36090]
:FORWARD ACCEPT [263:144982]
:OUTPUT ACCEPT [450:57803]
-A FORWARD -i usb0 -o local-network -j ACCEPT
COMMIT
# Completed on Sat Aug 17 20:50:44 2019
# Generated by xtables-save v1.8.2 on Sat Aug 17 20:50:44 2019
*nat
:PREROUTING ACCEPT [42:4401]
:INPUT ACCEPT [20:2013]
:POSTROUTING ACCEPT [0:0]
:OUTPUT ACCEPT [76:5665]
-A PREROUTING -i usb0 -p tcp -m tcp --dport 22 -j REDIRECT --to-ports 22
-A PREROUTING -i usb0 -p udp -m udp --dport 53 -j REDIRECT --to-ports 53
-A POSTROUTING -j MASQUERADE
-A POSTROUTING -s 10.1.1.0/24 ! -d 10.1.1.0/24 -j MASQUERADE
COMMIT
# Completed on Sat Aug 17 20:50:44 2019

also works with this iptables:

# Generated by xtables-save v1.8.2 on Sun Aug 18 16:29:53 2019
*filter
:INPUT ACCEPT [895:80551]
:FORWARD ACCEPT [5085:1730477]
:OUTPUT ACCEPT [766:85758]
COMMIT
# Completed on Sun Aug 18 16:29:53 2019
# Generated by xtables-save v1.8.2 on Sun Aug 18 16:29:53 2019
*nat
:PREROUTING ACCEPT [165:12567]
:INPUT ACCEPT [152:11771]
:POSTROUTING ACCEPT [0:0]
:OUTPUT ACCEPT [147:9966]
-A POSTROUTING -o wlan0 -j MASQUERADE
COMMIT
# Completed on Sun Aug 18 16:29:53 2019

ifconfig

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 15  bytes 9210 (8.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 15  bytes 9210 (8.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.1.1.1  netmask 255.255.255.0  broadcast 10.1.1.255
        inet6 fe80::25c:a702:1a70:ac6c  prefixlen 64  scopeid 0x20<link>
        ether fe:4d:37:95:9b:60  txqueuelen 1000  (Ethernet)
        RX packets 333  bytes 38248 (37.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 288  bytes 82047 (80.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.10.157  netmask 255.255.255.0  broadcast 10.10.10.255
        inet6 fe80::f94b:dcf:b227:5f43  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:b9:a3:4f  txqueuelen 1000  (Ethernet)
        RX packets 206  bytes 65212 (63.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 226  bytes 38265 (37.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lnxrouter -i usb0 works right! (iptables -F && iptables -t nat -F && systemctl stop dnsmasq && lnxrouter -i usb0)

but i need transparent proxy (SOCKS5 и SSH) and when i run: 
iptables -F && iptables -t nat -F && systemctl stop dnsmasq && ssh -fND 5555 root@167.71.50.214 && lnxrouter -i usb0 --tp 5555 --dns 1.1.1.1:853) there is no inet((((

I also tried pr0cks, followed all the instructions, but the Internet is the same as it was, but if you run curl ifconfig.me on rpi (connecting via ssh), then the Internet is proxied inside it ... I’ve been suffering for half a year !! ! please help !!!! (((((from me 'coffee' as you say))

garywill, 2019-08-20:
Hi there,

The port which iptables redirects to must be a transparent proxy (redsocks) port. We can't use socks port. `ssh -D` is a socks4 port (notice just 4). 
So if you use `ssh -D 5555`, then you need a program converting socks port to transparent proxy port, like
https://github.com/vi/tcpsocks
https://github.com/semigodking/redsocks
https://github.com/darkk/redsocks
https://github.com/nadoo/glider
https://github.com/snail007/goproxy
https://github.com/ginuerzh/gost
Seems that pr0cks is also for that, but I haven't ever tried it. 

I recommend you try gost, it's a converter supporting transparent tcp proxy and socks 4, and it provides binary file.
```
gost -L redirect://:1234 -F socks4://:5555 -D
```
use with `lnxrouter --tp 1234`.

Make sure the transparent proxy (port 1234 here) listens on `0.0.0.0` (Same to transparent DNS proxy)

Hope that help a bit.

closed 2019-08-22 by l4ckyguy

l4ckyguy, 2019-08-22:
THANK YOU VERY MATCH!!!111 it is wonderful, but DNS not work... i fix this by using dnscryptproxy + pr0cks. and i buy some coffee for you))) i am very sorry for my english, but can u help me with one more thing? how can i use socks4/5 with your script? i grab it by proxybroker, gost connecting is seccessful, but there is no internet. also i install gost to pizero, no support for armel(6). thank you for your time))) i'm from Russia :)

garywill, 2019-08-23:
>  how can i use socks4/5 with your script?

This script's `--tp` just uses iptables' `REDIRECT`, which needs transparent proxy (redsocks) port and doesn't support socks. So we **must use with other program** to use socks proxy. 

dnscryptproxy is ok, we can just use `lnxrouter --dns <dnscryptproxy port>`.

>  also i can't nstall gost to pizero, no support for armel(6)

goproxy and glide provide armv6 binary, but they doesn't seem to support socks4.

semigodking/redsocks and darkk/redsocks seems to support socks4, you can try compiling them. darkk/redsocks seems a well-known and traditional method. (I haven't tried them). Or, is there some program converting between socks 4 and 5?

> i grab it by proxybroker, gost connecting is seccessful, but there is no internet. 

Do you mean scanning socks proxies on Internet? Hmmmm..... I'm not sure if they're usable. Socks is not encrypted. ISP may ban the connection.

garywill, 2019-08-23:
I received your ETH. Thank you!

l4ckyguy, 2019-08-25:
so how can i use sock5 proxy?))))) i use gost as u say and it is work fine)) but i also want use socks)

garywill, 2019-08-26:
For example you have a socks proxy `a.a.a.a:1080` (`a.a.a.a` can be `127.0.0.1` or some public IP)

Run
```
gost -L redirect://:1081 -F socks://a.a.a.a:1080 -D
```
gost listens on port 1081

Use lnxrouter and redirect packets to port 1081
```
lnxrouter -i usb0 --tp 1081
```
OK socks is done.

----------------

As for DNS, there're different ways. 
( `--dns 1.1.1.1:853` is not going to work. 853 is DoT, but lnxrouter `--dns` requires regular DNS )

1. (recommended) Run dnscrypt-proxy listening on 5053. Use lnxrouter with `--dns 5053`

1. Don't serve DNS in LAN.  
Use lnxrouter with `--dhcp-dns 
1.1.1.1`. Let clients use public DNS (through socks).
That requires one of these conditions :
    1. Your socks 5 proxy supports UDP
    1. Clients supports TCP DNS

1. `--dns 1.1.1.1`
That will use `1.1.1.1` as LAN DNS's upstream (regular) DNS. 

1. Use your local ISP's DNS
Nothing extra to do

Note that 3 and 4 DNS queries don't go through proxy and not encrypted (!!).

-------------------

If you still have problems feel free to ask. Post what exactly you did and the logs.

l4ckyguy, 2019-08-28:
there is no internet, except lnxrouter --ap wlan0 666 --tp 1234 --dns 53 if dnscrypt-proxy listen 127.0.0.1:53 ((( also i need to use pr0cks because dns not changed(((( 
So i use this one: 
ssh -fND 1080 555.55.55.55 && pr0cks --dns-server 1.1.1.1:53 
gost -L redirect://:1234 -F socks://555.55.55:1080 -D
lnxrouter --ap wlan0 666 --tp 1234

another there is no internet and/or dns no change... 
when i bridge wifi with my gl-inet router (DNS-over-TLS with anti-dns-rebinding)  my dns is RIGHT. not isp Cloudflare but RIGHT, same as true IP ISP((( it is my dream for about six month to do same.... my girlfriend and my another girlfriiend go out)))) i mean you understand me

garywill, 2019-08-29:
Hi, there. Sorry but I think you didn't make yourself really clear.

The "pr0cks" is it [this](https://github.com/n1nj4sec/pr0cks) one?
Does your "use pr0cks" include manually adding iptables rules? Or running pr0cks script only? 
Do you want both the pi and clients go through proxy, or just clients?

> ((( also i need to use pr0cks because dns not changed((((

What you mean "dns not changed". Clients still use ISP's DNS? Or do you expect the pi's system DNS change?
lnxrouter's `--log-dns` may help. We need logs to debug.

Shouldn't  
`gost -L redirect://:1234 -F socks://555.55.55:1080 -D`
 be 
`gost -L redirect://:1234 -F socks://127.0.0.1:1080 -D` ?

l4ckyguy, 2019-08-29:
it is for example, bro)))) a do it!!! give me five, bro :D
pr0cks - real shit :D your script - the best!!))) gost -  also great!!  i love China :D yeah!! 
i use someone great framework, it is do wonderful and so easy. just 'same framework' gost and YOUR script do the things!!! yeah!! mail me and i say it name)) not here)) tssss))))))))))) thank you for your time. you are really help me!! thanks one more time!!!!!!!)))

l4ckyguy, 2019-08-29:
my mail)  1@l4cky.men or 1@l4cky.com :D

garywill, 2019-09-01:
Sent you a mail

Being busy these days....

merll002, 2025-05-03:
This is the weirdest exchange I have ever seen on github