garywill/linux-router #7: Error occurred ERROR: hostapd exited

doitranch, 2019-11-01:
I don't know if anyone is still active here but I cant seem to fix this... I'm running lnxrouter like: ./lnxrouter -n --ap ap0 SSID --password PASSKEY204 and it keeps breaking after saying "lnxrouter has been setup" I don't know what information to include other than I'm running this on a Raspberry Pi Zero W with Raspbain Lite. Any help is appreciated.

garywill, 2019-11-02:
Hi.

>  from a Non Internet Local AP

Does it work well when privide Internet?

Have you tried other method to make AP? Like create_ap?

Check if hostapd's process normal (running? zombie?). You can use `lnxrouter --hostapd-debug 1` to show hostapd's debug log. Post the log with your whole command so that people may find the reason.

doitranch, 2019-11-02:
Ok will do but I'm out of town at the moment.

doitranch, 2019-11-03:
Thanks for the help hostapd was dead

closed 2019-11-03 by doitranch