garywill/linux-router #8: how to route TO/FROM local LXD bridge interface and local VPN interface

bmullan, 2020-02-07:
Just found _linux-router_  tool and it looks very promising for a lot of things.

I have a question on use though.   Many VPN's only implement a TUN device (TUN - TCP vs TAP - ethernet).

If I were using WireGuard or Nebula can your tool enable traffic from a NODE's  lxdbr0 bridge interface (LXD) to/from the VPN interface.

**example:**
using Wireguard it creates an interface on the NODE called "wg0"
or if using Nebula it creates an interface on the NODE called "nebula1"

I was hoping the following might work for wireguard if configured on NODE1 and NODE1 (both ubuntu 18.04

**$ sudo ./linux-router.sh -i lxdbr0 -o wg0 --daemon
$ sudo ./linux-router.sh -i wg0 -o lxdbr0 --daemon**

But from NODE1 and **_I am not able to ping_** NODE2's lxdbr0 interface although I can ping the WireGuard interface (wg0) on NODE2.

Since it looks like you've used LXD I thought I'd ask how this might be done with your linux-router?

thanks for any information 

brian

garywill, 2020-02-08:
Hi there,

This script supports only NAT, so, VPN and LXD's network are two different networks. However you can get two networks communicate by manually adding an item in routing table.

If you want to truely "bridge" two interfaces, this script can't  (for now).

I assume this is as you described:
```
(run on NODE2)
lnxrouter -i lxdbr0 -o wg0 -g 192.168.20.1
(that's enough. no need to use two lnxrouter command)
```
```
NODE1-------(wg)-----------NODE2-----(lnxrouter)-----LXD containers
192.168.10.2        192.168.10.1
                      192.168.20.1                192.168.20.2
```
Now in lnxrouter network can ping wg network. But in wg network can't ping lnxrouter network.

On NODE1 run
```
sudo ip route add 192.168.20.0/24 via 192.168.10.1
```
Now NODE1 can ping lxdrouter network.

bmullan, 2020-02-08:
Actually both NODE1 and NODE2 run LXD and both have containers.

I want to ping "from" a container on NODE 1 "to" a container on NODE2.

garywill, 2020-02-08:
That can be accomplished by many different ways. I think you are looking for a general network solution. 

Assuming you just want the two containers in two nodes to communicate, do one more set of symmetry commands as I described above, modify the IP yourself.

If you still have question, please tell us how much you know about network and routing, and how familiar you are with Linux and its basic networking tools and LXD. So people know how to help.

And as I remember LXD itself seemed to have NAT, in that case linux-router is not necessary. 
And, simply using Linux command to  bridge `lxdbr` and `wg` may do the job (if wg protocol supports bridging).

bmullan, 2020-02-08:
@garywill 
In regards to..
> _And, simply using Linux command to bridge lxdbr and wg may do the job (if wg protocol supports bridging)_

I had mentioned in the original post that WG creates a TUN (Layer 3) interface so it and the LXDBR0 bridge (Layer 2) just can't just be "bridged" together.

Also, I've used LXD for years.

I think my knowledge of routing/switching is pretty good.  

_My original problem just stems from not understanding your README and tool & what it can and can not do._ 

One example in the README had me puzzled as to what it meant:

> _Provide an interface's Internet to another interface_
> 
> # lnxrouter -i eth1 -o vpn0 --dhcp-dns 1.1.1.1

Since _eth0 is more frequently_ the Internet interface (unless your machine is wifi connected)  I had "_assumed_" that the README saying "*provides an interfaces internet**" just only meant something like "_connects the traffic from segment behind eth1_" -to- the output  "_-o vpn0_".   

My use-case is a multi-tenant, multi-node/host/server, multi-cloud environment using LXD.   The following will give you a general idea:

Click on the image to enlarge:

[image]

I've got several **_"working" solutions to my "use-case_**" _already_ but I'm trying to come up with a couple more to document before I open them up for others.

One uses a full-mesh auto-learning vpn (not wireguard), and BGP w/BGP Route Reflectors configured.

The other uses WireGuard, VxLAN and BGP w/BGP Route Reflectors configured.

Both work well, but I have a 3rd in mind and just keep looking for interesting tools to help.

closed 2020-02-08 by bmullan