garywill/linux-router #9: Raspberry Pi 3B AP without internet + Android 10 Phone

tomicooler, 2020-03-29:
Hi,

i'm trying to achieve something with my Raspberry Pi 3B and an Android phone. Unfortunately I'm stuck. I think the real problem is the Android 10's routing, because the AP that I create on my Raspberry works with a laptop. Anyway I try to summarize what I've learned about the problem, and maybe you might come up with a solution/workaround.

First, I have a WiFi OBD (on board diagnostic) that creates a network.

```
  gateway: 192.168.0.10, subnet 255.255.255.0
  dns: 192.168.0.10
```

ip route show (on my phone)
```
192.168.0.0/24 dev wlan0 proto kernel scope link src 192.168.0.19
```

If I connect to this network with my phone, I must turn off the mobil data first, then I can ping 192.168.0.10 and I also able to communicate with the obd interface. Unfortunately If I connect more devices, the devices can't ping each other just the gateway. So my idea: raspberry connects to the Wifi_OBDII network and simultaneously creates an AP. My service on pi can communicate with the obd interface, and my phone can communicate with the pi. The phone doesn't need to access the obd interface.

More info on WiFi_OBDII (from my laptop)
```
Nmap scan report for 192.168.0.10
Host is up (0.018s latency).
Not shown: 999 filtered ports
PORT      STATE SERVICE
30000/tcp open  ndmps

$ip a
3: wlp1s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 34:f3:9a:a3:ee:36 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.20/24 brd 192.168.0.255 scope global dynamic noprefixroute wlp1s0
       valid_lft 86343sec preferred_lft 75543sec
    inet6 fe80::36f3:9aff:fea3:ee36/64 scope link 
       valid_lft forever preferred_lft forever

$ip route show
default via 192.168.0.10 dev wlp1s0 proto dhcp src 192.168.0.20 metric 303 
192.168.0.0/24 dev wlp1s0 proto dhcp scope link src 192.168.0.20 metric 303
```

Details about how I tried to use linux-router.

```
./lnxrouter -n --ap wlan0_ap TomiMetria --hostname tomimetria -g 192.168.27.10
WARN: brmfmac driver doesn't work properly with virtual interfaces and
      it can cause kernel panic. For this reason we disallow virtual
      interfaces for your adapter.
      For more info: https://github.com/oblique/create_ap/issues/203
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
PID: 2858
Target interface is wlan0_ap
WARN: Wifi is not protected by password
haveged_watchdog PID: 2895

Starting hostapd
hostapd PID: 2898
Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0_ap.conf.Ecl/hostapd.conf
wlan0_ap: Could not connect to kernel driver
Using interface wlan0_ap with hwaddr b8:27:eb:fb:8e:2c and ssid "TomiMetria"
wlan0_ap: interface state UNINITIALIZED->ENABLED
wlan0_ap: AP-ENABLED 
No Internet sharing

iptables: allow DNS port access
ACCEPT  tcp opt -- in wlan0_ap out *  192.168.27.0/24  -> 192.168.27.10   tcp dpt:53 /* lnxrouter-2858-wlan0_ap */
ACCEPT  udp opt -- in wlan0_ap out *  192.168.27.0/24  -> 192.168.27.10   udp dpt:53 /* lnxrouter-2858-wlan0_ap */

iptables: allow DHCP port access
ACCEPT  udp opt -- in wlan0_ap out *  0.0.0.0/0  -> 0.0.0.0/0   udp dpt:67 /* lnxrouter-2858-wlan0_ap */

Starting dnsmasq
Mar 29 16:05:24 dnsmasq[2936]: started, version 2.80 cachesize 150
Mar 29 16:05:24 dnsmasq[2936]: compile time options: IPv6 GNU-getopt DBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
Mar 29 16:05:24 dnsmasq[2936]: warning: no upstream servers configured
Mar 29 16:05:24 dnsmasq-dhcp[2936]: DHCP, IP range 192.168.27.10 -- 192.168.27.250, lease time 1h
Mar 29 16:05:24 dnsmasq-dhcp[2936]: DHCP, sockets bound exclusively to interface wlan0_ap
Mar 29 16:05:24 dnsmasq[2936]: cleared cache
dnsmasq PID: 2936

== Setting up completed, now linux-router is working ==
wlan0_ap: STA a0:28:ed:85:00:33 IEEE 802.11: associated
wlan0_ap: AP-STA-CONNECTED a0:28:ed:85:00:33
wlan0_ap: STA a0:28:ed:85:00:33 RADIUS: starting accounting session 2FD2F153B87501C5
Mar 29 16:06:14 dnsmasq-dhcp[2936]: DHCPDISCOVER(wlan0_ap) a0:28:ed:85:00:33 
Mar 29 16:06:14 dnsmasq-dhcp[2936]: DHCPOFFER(wlan0_ap) 192.168.27.204 a0:28:ed:85:00:33 
Mar 29 16:06:14 dnsmasq-dhcp[2936]: DHCPREQUEST(wlan0_ap) 192.168.27.204 a0:28:ed:85:00:33 
Mar 29 16:06:14 dnsmasq-dhcp[2936]: DHCPACK(wlan0_ap) 192.168.27.204 a0:28:ed:85:00:33 
wlan0_ap: STA 00:28:f8:4b:af:dc IEEE 802.11: associated
wlan0_ap: AP-STA-CONNECTED 00:28:f8:4b:af:dc
wlan0_ap: STA 00:28:f8:4b:af:dc RADIUS: starting accounting session 41CE24EFC36B8576
Mar 29 16:18:37 dnsmasq-dhcp[2936]: DHCPDISCOVER(wlan0_ap) 00:28:f8:4b:af:dc 
Mar 29 16:18:37 dnsmasq-dhcp[2936]: DHCPOFFER(wlan0_ap) 192.168.27.181 00:28:f8:4b:af:dc 
Mar 29 16:18:37 dnsmasq-dhcp[2936]: DHCPREQUEST(wlan0_ap) 192.168.27.181 00:28:f8:4b:af:dc 
Mar 29 16:18:37 dnsmasq-dhcp[2936]: DHCPACK(wlan0_ap) 192.168.27.181 00:28:f8:4b:af:dc tomi-pc

[root@alarmpi linux-router]# ./lnxrouter  --list-clients wlan0_ap
MAC                  IP                 Hostname
00:28:f8:4b:af:dc    192.168.27.181     tomi-pc
a0:28:ed:85:00:33    192.168.27.204     LICENSE
```
192.168.27.204 is my phone. 192.168.27.181 is a laptop. The laptop works, I mean from the laptop I can ping 192.168.27.10 the gateway.

```
[root@alarmpi linux-router]# arp -a
? (192.168.27.204) at a0:28:ed:85:00:33 [ether] on wlan0_ap
_gateway (192.168.1.1) at 48:f8:b3:14:89:e6 [ether] on wlan0
? (192.168.27.181) at 00:28:f8:4b:af:dc [ether] on wlan0_ap
? (192.168.1.102) at 34:f3:9a:a3:ee:36 [ether] on wlan0
```

I can ping the phone from the pi
```
[root@alarmpi linux-router]# ping 192.168.27.204
PING 192.168.27.204 (192.168.27.204) 56(84) bytes of data.
64 bytes from 192.168.27.204: icmp_seq=6 ttl=64 time=50.6 ms
```

I can ping the phone from the connected laptop as well.

But I can't ping the gateway from the phone.

phone details
```
$ ip route show
192.168.27.0/24 dev wlan0 proto kernel scope link src 192.168.27.204
```
I can ping the connected laptop from the phone, but I can't ping that damned gateway.

This is how I created the wlan0_ap on the pi.
```
$ iw dev wlan0 interface add wlan0_ap type __ap
$ macchanger -e wlan0_ap

[root@alarmpi linux-router]# iw wlan0_ap info
Interface wlan0_ap
	ifindex 4
	wdev 0x2
	addr b8:27:eb:fb:8e:2c
	ssid TomiMetria
	type AP
	wiphy 0
	channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
	txpower 31.00 dBm

[root@alarmpi linux-router]# iw wlan0 info
Interface wlan0
	ifindex 3
	wdev 0x1
	addr b8:27:eb:16:1b:4c
	ssid LinkSYS
	type managed
	wiphy 0
	channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
	txpower 31.00 dBm
```

Anyway, since the WiFi_OBDII creates a network where the gateway is accessible from my android phone, there must be a way to achieve this, isn't it?

garywill, 2020-04-03:
Hi there,
Thanks for feedback.
These are what I guess you could try (and hmmm may not help):
1. I saw you use `./lnxrouter -n` . The right way to do is don't use `-n`, so on pi `wlan0` and `wlan0_ap` can forward each other's traffic.
1. 
    > phone details
    $ ip route show
    192.168.27.0/24 dev wlan0 proto kernel scope link src 192.168.27.204

    There should also be `default via 192.168.27.10` so the phone knows where to reach the gateway. Is it there on phone's routing table?
1. I guess more likely it's the OBD's problem (just guess)
1. 
    > Pinging the phone from the laptop works, but the time varies 20-400 ms.

    If the phone's screen is off, the phone's wifi may go in powersaving mode.
1. Do you have more device? Test more device. If always fail, try debugging with Wireshark or tcpdump

tomicooler, 2020-04-05:
Hi,

1. no luck without the -n.
2. no there wasn't a default route, maybe that was the problem. However, I don't have root on my phone, so I don't think I could have fix that.

Well, I eventually got a decent workaround for this. I bought a WiFi adapter, so I have 2 interface on my Pi (wlan0 the internal, wlan1 the wifi adapter). I make the hotspot with my phone, the Pi connects to this network with the wlan0 interface (static ip) (because it is faster, and between the Pi <-> Phone there is a bigger traffic, like a screen sharing..), and the wlan1 connects to the OBD's wifi.

The wifi adapter is a Tenda U3 (quite a cheap one), but it said it can do Soft-AP. My original plan was to create the hotspot on the Pi with this Tenda U3, but as I found out Soft-AP is not what I thought it was. They state in their documentation:

> Adapter and SoftAP. SoftAP can transfer your computer's wired network into wireless signal. SoftAP is working only when PC could access the internet.

As far as I remember when I made the virtual wlan0_ap interface on the Pi, and the wlan0 was connected to my home wifi network (which had internet access as well), everything was working with one interface on the Pi. Maybe due to similar reasons that this SoftAP means.

Anyway. Thanks for the help! I close the issue.

closed 2020-04-05 by tomicooler