garywill/linux-router #12: Error Device or resource busy

Gictorbit, 2021-02-06:
an error occurred while bringing the virtual interface up
command:
```
sudo ./lnxrouter --ap wlp0s20f3 victorAP -p 12345678
```
output:
```
PID: 213324
Target interface is wlp0s20f3
Use random LAN IPv4 address 192.168.238.1
Creating a virtual WiFi interface... 
x0wlp0s20f3 created)
Set x0wlp0s20f3 unmanaged by NetworkManager
Assigning MAC address 5e:a8:4c:ab:73:5a to virtual interface x0wlp0s20f3 according to wlp0s20f3 ...
RTNETLINK answers: Device or resource busy
Error occured

ERROR: Failed bringing x0wlp0s20f3 up

Doing cleanup.. 
iptables: stop NAT
iptables: unallow DNS
iptables: unallow dhcp
Exiting: This is the only running instance
Cleaning up done

```

garywill, 2021-02-06:
Hi there, 
thanks for feedback.

The log shows `5e` was the first byte of the wifi interface, which was abnormal. device manufacturers' MAC address' first byte should always be multiple of 4.

I found sometimes creating virtual wifi interface causes the real wifi interface turning into a random strange MAC address. I'm still confused why that happends (so I added a delay at https://github.com/garywill/linux-router/blob/2341938b762005230fef8199c478d462f6f9cfc7/lnxrouter#L1598 sometimes the MAC address goes back again automatically)

Please check what `wlp0s20f3`'s original MAC address is. For example `macchanger` can do that:
```
macchanger -s wlp0s20f3
```
it will return something like
```
Current MAC:  xxxxx
Permanent MAC: xxxxx
```

Try these:
1.   If MAC changed, use `macchanger -p wlp0s20f3` to set it back then try again.
    If still fail, try increasing the sleep time

2. `rfkill unblock wifi` before running `lnxrouter`

3. try with `--no-virt`
4.   are you using the latest commit?
5. does create_ap work?

Gictorbit, 2021-02-06:
> I found sometimes creating virtual wifi interface causes the real wifi interface turning into a random strange MAC address. I'm still confused why that happends (so I added a delay at

hmm, that's weird. also I tested `create_ap` on the same device using same linux distribution and it worked fine without any problem

> Please check what `wlp0s20f3`'s original MAC address is. For example `macchanger` can do that:

here is the output of `macchanger -s wlp0s20f3`
```
Current MAC:   92:cc:2a:xx:xx:xx (unknown)
Permanent MAC: 80:32:53:xx:xx:xx (unknown)
```

> 4\. are you using the latest commit?

yes
and create_ap was ok and worked great

garywill, 2021-02-07:
> I tried this method and works fine without creating virtual interface but sometimes we really need to set a virtual interface up
> increased sleep time to 6 and error occurred again

When creating virtual interface, is the error happening always or sometimes?

NetworkManager is a possible reason. If NM running, it will give the new virtual interface a random MAC. MAC will go back after setting NM unmanaged. But why `wlp0s20f3` too? Wired. Only the virtual interface should go a random MAC.

Try increasing this sleep time (after setting NM unmanaged)
https://github.com/garywill/linux-router/blob/2341938b762005230fef8199c478d462f6f9cfc7/lnxrouter#L778

Gictorbit, 2021-02-07:
> When creating virtual interface, is the error happening always or sometimes?

This error always happens, but it works fine with `create_ap` I think it is a bug maybe

> Try increasing this sleep time (after setting NM unmanaged)

I tried this,it doesn't help. same error occurs. I increased sleep to 6 second in line 778 and here is the output:
```
[sudo] password for victor:          
PID: 194580
Target interface is wlp0s20f3
Use random LAN IPv4 address 192.168.197.1
Creating a virtual WiFi interface... 
x0wlp0s20f3 created)
Set x0wlp0s20f3 unmanaged by NetworkManager
Assigning MAC address 42:6d:9b:07:9e:b4 to virtual interface x0wlp0s20f3 according to wlp0s20f3 ...
RTNETLINK answers: Device or resource busy
Error occured

ERROR: Failed bringing x0wlp0s20f3 up

Doing cleanup.. 
iptables: stop NAT
iptables: unallow DNS
iptables: unallow dhcp
Exiting: This is the only running instance
Cleaning up done

```

garywill, 2021-02-07:
I did some test with
```
watch -n 0.1 ip a
```
I found every time I check/uncheck the "enabe wifi" on NM, my `wlan0` MAC changes. 
Having NM wifi on, runing `rfkill (un)block wifi` also makes MAC random.

Recently I upgrade Linux last month. Then the strange MAC behavior began. Did you upgrade recently? I'm using 
- NetworkManager 1.22.10
- util-linux-2.33.1
- kernel 5.3.18
- KDE plasma5-session 5.18.6

create_ap writes NM's config file to set unmanaged **before** creating virtual interface.
linux-router uses `nmcli` to set **after** creating.

Could you try 
```
sudo systemctl stop NetworkManager
```
(after having connected to your Internet Wifi)
then run `lnxrouter`

Gictorbit, 2021-02-07:
> found every time I check/uncheck the "enabe wifi" on NM, my `wlan0` MAC changes.
> Having NM wifi on, runing `rfkill (un)block wifi` also makes MAC random.

Interesting. I use arch Linux. Since it is rolling release, the packages are updated to the last version. the packages specs are below
+ NetworkManager 1.28.1
+ util-linux 2.36.1
+ kernel 5.10.8
+ Gnome 3.38.3

> create_ap writes NM's config file to set unmanaged **before** creating virtual interface.
> linux-router uses `nmcli` to set **after** creating

hmm, so that is why `create_ap` works and `linux-router` encounters error

> Could you try
> 
> ```
> sudo systemctl stop NetworkManager
> ```
> 
> , reset everything, then run `lnxrouter`

I did it then I restart `NetworkManager` service and I ran` lnxrouter` command, same error happened

fmichaut-dev, 2021-07-18:
Hello, just wanted to add that after upgrading to Fedora 32, create_ap doesn't work anymore (the ap is created and accessible, but the device is stuck in the connection status and eventualy timeout)

I tired linux-router to see if it would fix the issue, but I run into the same error with a virtual interface : `RTNETLINK answers: Device or resource busy`

Is there a fix planned for this issue ?

garywill, 2021-07-18:
@fmichaut-dev 

> after upgrading to Fedora 32, create_ap doesn't work anymore (the ap is created and accessible, but the connecting device is stuck in the connection step and eventualy timeout)

Other Fedora users report similar issue https://github.com/garywill/linux-router/issues/19#issuecomment-873337509
Currently in my mind is that we need someone to disable system services one by one to figure out which service is causing problem. 

------------

> I tired linux-router to see if it would fix the issue, but I run into the same error with a virtual interface : RTNETLINK answers: Device or resource busy

 I said in above posts:

> create_ap writes NM's config file to set unmanaged **before** creating virtual interface.
linux-router uses nmcli to set **after** creating.

So I guess it's the NetworkManager causing that.

So could someone encountering `RTNETLINK answers: Device or resource busy` try disabling NM then run?

-------

Sorry for these issues but I've been busy on other jobs (still need 1-2 months. or even longer). That's why we need more people for this `linux-router`

fmichaut-dev, 2021-07-18:
> So could someone encountering `RTNETLINK answers: Device or resource busy` try disabling NM then run?
> 

So I just tried that on my fedora and it allows `linux-router` to start and I can connect with another device.
But I fall back to the same issue of device timing out trying connecting to the ap like in the other issue you linked.

garywill, 2021-12-09:
I added some delays and log gathering code in [branch dev-12](https://github.com/garywill/linux-router/tree/dev-12)

@ThomasK7775  Could you who encounter this issue do some tests and show the results for us? I can't do it myself cause I can't reproduce on my machine.

1. Run the `master` branch with `--no-virt`:
    ```bash
    lnxrouter --ap wlan0 wifi -p 12345678 --no-virt
    ```
    (according to users reported above, this worked. I just want to make sure it still works now)

2. Run the `dev-12` branch (e1ccfac) and record the logs 
3. Comment out this line then run again:
    https://github.com/garywill/linux-router/blob/e1ccfac40cc969863a3bc0581409ce286eed837d/lnxrouter#L1656
4. Disable MAC randomization during Wi-Fi scanning (restart NM) then run:
    ```
    /etc/NetworkManager/conf.d/wifi_rand_mac.conf

    [device]
    wifi.scan-rand-mac-address=no
    ```
    (https://wiki.archlinux.org/title/NetworkManager#Configuring_MAC_address_randomization)

Thanks in advance. We need tests and info cause still haven't got idea.

DeadSix27, 2022-01-02:
Recent log on dev-12 branch:
https://gist.github.com/DeadSix27/1a6ecc56f2e8addbe1eacae666d2efbe

Facing the same issues as outlined above

garywill, 2022-01-03:
I opened an issue on NetworkManager: https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/886

They require more verbosed logs.

1. I updated  log-gathering code in [dev-12](https://github.com/garywill/linux-router/tree/dev-12) branch (204756c)
2. Before you start the log-gathering, restart `wpa_supplicant` with `-dd` to get more log

You can post more logs here, or would rather directly submit the logs to them.

Wizard254, 2022-02-07:
Run into this error.
Setup: Kubuntu 21.04

[image]

Solved this by deactivating wireless connections, (DO NOT TURN ON AIRPLANE MODE), then running the command. 
The hotspot is created successfully. Then because the wireless connection was disabled, enable it again. Then connect to the target WiFi, and the internet connection should be passed through to the hotspot devices.

Note that I'm using the same, inbuilt WiFi adapter.

From the above, you can only create a Hotspot first, then connect the same interface to an internet connection second.

babam86, 2022-09-12:
create_ap worked fine on me (Intel wifi 6 ax201)
Using lnxrouter must use these steps for it to work
`nmcli radio wifi off
lnxrouter --ap wlan0 MyAccessPoint -p MyPassPhrase &
sleep 6
nmcli radio wifi on`

TheGrave, 2024-07-01:
> `nmcli radio wifi off`

This doesn't work in Ubuntu 24.04. nmcli radio wifi off puts the machine in Airplane mode according to network manager and lnxrouter fails:

`linux-router 0.7.6b (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 254671
Target interface is wlp1s0 (9e:01:81:11:c3:49) 
PCI: 0000:01:00.0
System-already-loaded driver: mt7921e
 Network controller [0280]: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter [14c3:0616] (rev 01)

Use random LAN IPv4 address 192.168.196.1
Creating a virtual WiFi interface... 
x0wlp1s0 created
Channel not specified, use default
Error occured

ERROR: Your adapter can not transmit to channel 1, frequency band 2.4GHz.

Doing cleanup.. 
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done
`

osamwelian3, 2025-04-12:
> `create_ap` worked fine on me (Slackware 15.0 - Intel Wi-Fi 6 AX201) Using `lnxrouter` must use these steps for it to work
> 
> ```
> nmcli radio wifi off
> lnxrouter --ap wlan0 MyAccessPoint -p MyPassPhrase &
> sleep 6
> nmcli radio wifi on
> ```
> 
> If I use `--no-virt` then the wifi is disconnected/disabled so I can't share the internet connection.

Interestingly this worked for me on Ubuntu 24.04.1 LTS.

f-michaut, 2025-08-02:
Hi, issue persists on Fedora 42.

```
Creating a virtual WiFi interface... 
x0wlp58s0 created
Freq band: 2.4 GHz   Channel: 13
Set x0wlp58s0 unmanaged by NetworkManager
Assigning MAC address ce:35:f0:da:a4:b0 to virtual interface x0wlp58s0 according to wlp58s0 ...

iptables v1.8.11 (legacy)
Notice: Not showing all operations done to iptables rules
RTNETLINK answers: Device or resource busy
Error occured

ERROR: Failed bringing x0wlp58s0 up
```

The weird thing is, if I run :
```
sudo systemctl stop NetworkManager
sudo ./lnxrouter --ap wlp58s0 my-ssid -p my-password
# Wait for lnxrouter to display == Setting up completed, now linux-router should be working ==
sudo systemctl start NetworkManager
```
Then the AP is created sucefully, and I can connect to it has internet access.

I would expect that when `lnxrouter` says `Set x0wlp58s0 unmanaged by NetworkManager` that NetworkManager would not be a problem, but looks like it still is ?
Or is the `RTNETLINK answers: Device or resource busy` related to `wlp58s0` instead of `x0wlp58s0` ?