garywill/linux-router #16: Wifi hotspot without internet

SomaGallai, 2021-03-26:
Hi,

I am trying to create a wifi hotspot, that works without internet connection and still I would be able to ssh to it from another computer. At the moment it is connected to another router and the hotspot is working as expected, but when there is no internet the SSH is not working.

thisisagan, 2021-12-16:
metoo

garywill, 2021-12-16:
Could you guys be more clear? I dont understand whats going on

aiden-loroco, 2022-01-24:
I would also like this. The use case is we have an rpi running with no touch screen or user input device. If the box can't connect to the wifi it knows about, then we'd like it to turn into an ap that serves just a simple ui which the user can connect to (on their phone for example) and supply the local wifi ssid and pw. I'm running:

```
sudo ./lnxrouter --ap wlan0 test-hello -p test-hello --no-dns --no-dnsmasq -w 2
```

I can see the network on my phone, but it fails to login. Note that I added --no-dns as I was seeing dnsmasq errors, and i'm not sure i need dns for my application.

SomaGallai, 2022-02-01:
Sorry for the inactivity, the issue was solved by creating a configuration file with all the settings that can be seen in the 'define_global_variables()' section of your software and a small script that does the following: 

- shutdown wifi
- enable the device/port wished to provide the AP
- service start create_ap

Hope this provide some help people still trying to achieve

closed 2022-02-01 by SomaGallai