garywill/linux-router #18: iptables: Couldn't load match `comment':No such file or directory

Ale32bit, 2021-04-02:
Upon running `sudo lnxrouter --ap wlan0 Alex -p #### ` iptables fails.

```
WARN: brmfmac driver doesn't work properly with virtual interfaces and
      it can cause kernel panic. For this reason we disallow virtual
      interfaces for your adapter.
      For more info: https://github.com/oblique/create_ap/issues/203
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
PID: 3969
Target interface is wlan0 (dc:a6:32:ae:05:4b)
Use random LAN IPv4 address 192.168.77.1
haveged_watchdog PID: 4041

Starting hostapd
hostapd PID: 4044
Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.ycq/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:ae:05:4b and ssid "Alex"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED

iptables: NAT
iptables v1.8.2 (nf_tables): Couldn't load match `comment':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
Error occured

Doing cleanup..
Terminated
Killed 4041 lnxrouter
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Killed 4044 hostapd
iptables: stop NAT
iptables: unallow DNS
iptables: unallow dhcp
Exiting: This is the only running instance
Cleaning up done
```

garywill, 2021-04-03:
> iptables v1.8.2 (nf_tables): Couldn't load match `comment':No such file or directory

Your iptables is nftables backend. I haven't used or tested with it. Am using legacy iptables.

You can try modify
https://github.com/garywill/linux-router/blob/1e3c5004c307cd6ba23aaddf401c72ee5c26b81b/lnxrouter#L796-L805
delete `-m comment --comment "lnxrouter-$$-$SUBNET_IFACE"`

If still see problem, try use legacy iptables. For example switch to legacy like
https://github.com/oblique/create_ap/issues/373
https://github.com/oblique/create_ap/issues/433

garywill, 2021-04-03:
Before nftables support plan maded, I'm considering an option to specify iptables legacy path

Ale32bit, 2021-04-03:
Tried switching to iptables-legacy and the error still happens

```
iptables: NAT
iptables v1.8.2 (legacy): Couldn't load match `comment':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.
Error occured
```

Am I missing something?

garywill, 2021-04-03:
> Couldn't load match `comment':No such file or directory

your iptables doesn't support comment.
Delete `-m comment --comment "lnxrouter-$$-$SUBNET_IFACE"` (see above said)

What OS are you using? What version?

garywill, 2021-04-17:
Added a comment module judgement

BenjaminWegener, 2022-06-11:
```
root@localhost:/usr/local# NETFILTER_XT_MATCH_COMMENT=0 /snap/bin/lxd init --auto
Error: Failed to create local member network "lxdbr0" in project "default": Failed to setup firewall: Failed to run: iptables -w -t nat -I POSTROUTING -s 10.74.220.0/24 ! -d 10.74.220.0/24 -j MASQUERADE -m comment --comment generated for LXD network lxdbr0: iptables v1.8.4 (legacy): Couldn't load match `comment':No such file or directory
```

`

closed 2022-09-22 by Ale32bit