garywill/linux-router #24: Request for clarification regarding OpenWrt

sheilbronn, 2021-09-18:
Hi! This is more a request for clarification rather than filing an concrete issue:
Will linux-router also work with OpenWrt?

garywill, 2021-09-18:
Hi there! Thanks for question. This is the first time we hearing from OpenWrt user!

Our script has been used by many computer and RPi users. As for router devices, I myself have (only) one running Linux. It has a broadcom chip. masquerade/forwarding is not handled by Linux, but by hardware. So no `iptables`. And it hasn't `hostapd`, wifi is handled by hardware too. So I've never tested this script on that router.

Though, router devices have different models. 
Looking forward to hearing other router device users' feedback!

sheilbronn, 2021-09-18:
So you're saying that - since OpenWrt has iptables and hostapd - I should be giving it a try?!

garywill, 2021-09-18:
Yes, why not? Just one command. Make sure meet the required dependencies

atrent, 2021-09-18:
What would be the purpose of linux-router on a... router?

garywill, 2021-09-19:
I think I didn't make clear enough on yesterday's reply.

Having `hostapd` and `iptables` installed is not the key. 
Key is: Hardware or software mode the device is 

If one router
- is a "software router"
- or is a "hardware router" which can also be set to "software" mode

then I believe 99% there's chance we can make our script work.

To check S or H, we can check:
- use `tcpdump` to see if all traffics passing the router physically reach Linux or not
  (not only those traffic from/to router itself, eg our SSH connection from PC to router)
- is iptables (or nft) empty or not
- is a `hostapd` process running when wifi's on
- ......

That's not all. Router may have something set by default, eg VLANs, which not taken into account by our script yet. (So welcome users' feedback)

> What would be the purpose of linux-router on a... router?

Hmmm... maybe our transparent proxy feature?

sheilbronn, 2021-09-19:
> 
> 
> What would be the purpose of linux-router on a... router?

... easily added additional SSID's...

sheilbronn, 2021-09-19:
> 
> 
> I think I didn't make clear enough on yesterday's reply.
> 
> ...
> If one router
> 
>     * is a "software router"
> 
>     * or is a "hardware router" which can also be set to "software" mode
> 
> then I believe 99% there's chance we can make our script work.
> 
> To check S or H, we can check:
> 
>     * use `tcpdump` to see if all traffics passing the router physically reach Linux or not
>       (not only those traffic from/to router itself, eg our SSH connection from PC to router)
> 
>     * is iptables (or nft) empty or not
> 
>     * is a `hostapd` process running when wifi's on
> 
>     * ......
> 
I wanted to try it on a Freifunk node (based on Gluon, based on OpenWrt, hardware is TP-Link TL-WR810N) to add an additional SSID.
The node (being based on OpenWrt) has a non-empty iptables (iptables --list), a running hostapd process and all traffic is encrypted before being tunneled through a Wireguard VPN. So I assume strongly, that the kernel sees all trffic....

atrent, 2021-09-20:
> > What would be the purpose of linux-router on a... router?
> 
> ... easily added additional SSID's...

this is a standard (and easily configured via luci or config files) feature of OpenWrt

sheilbronn, 2021-09-20:
> > > What would be the purpose of linux-router on a... router?
> >  
> > ... easily added additional SSID's...
> 
> this is a standard (and easily configured via luci or config files) feature of OpenWrt

Yes, regarding OpenWrt you're right, but maybe not for Gluon and/or Freifunk (which I would like try out as soon as I have another device to experiment with)

closed 2024-12-30 by garywill