garywill/linux-router #27: Why when I stop wifi hotspot in python command line, python will be teminated too?

VioletL-in, 2021-11-06:
it really confuse me, does anyone have met the same problem?

other information may be helpful:
 - Ubuntu 20.04.3 LTS (GNU/Linux 5.4.0-89-generic x86_64)
 - Python 3.8.10

garywill, 2021-11-06:
This is how linux-router kill processes (doing cleanup).
https://github.com/garywill/linux-router/blob/master/lnxrouter#L1223-L1226

This is what on my system when running lnxrouter:
```
$ sudo lnxrouter -i wlan0
[sudo] root 的密码：
linux-router 0.6.5 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 12497
Target interface is ..........

................

Doing cleanup.. 
Nov  6 17:39:15 dnsmasq[12646]: exiting on receipt of SIGTERM
Killed dnsmasq.pid 12646 dnsmasq
Killed keep_running.pid 12684 bash
Undoing iptables changes ..
Restore wlan0 managed by NetworkManager
Exiting: This is the only running instance
Killing PGID 12496 ...
Cleaning up done
```

```
$ ps axfj
 2826  4938  4938  4938 pts/10   12496 Ss    1000   0:00  |   \_ bash
 4938 12496 12496  4938 pts/10   12496 S+       0   0:00  |   |   \_ sudo lnxrouter -i wlan0
12496 12497 12496  4938 pts/10   12496 S+       0   0:00  |   |       \_ /bin/bash /home//bin/lnxrouter -i wlan0
12497 12581 12496  4938 pts/10   12496 S+       0   0:00  |   |           \_ cat /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.w7uJtx/dnsmasq.log
12497 12684 12496  4938 pts/10   12496 S+       0   0:00  |   |           \_ bash -c while :; do sleep 8000 ; done 
12684 12685 12496  4938 pts/10   12496 S+       0   0:00  |   |               \_ sleep 8000
```

Maybe we're doing in the wrong way? I assumed lnxrouter ran by a `sudo` which would no longer be needed.

Could you provide your python code, your log, and your `ps axfj` so we can test.

VioletL-in, 2021-11-09:
Thanks for your reply, here is my python code(`manage_ap.py`):
```
import os
import time

from subprocess import run, Popen, DEVNULL

interface_to_create_ap = 'wlan0'
interface_to_internet = 'ens0'
tool_path = './lnxrouter'

def status():
    ps = run(f"{tool_path} --list-running".split(' '), capture_output=True, encoding='utf-8')
    if interface_to_create_ap in ps.stdout:
        return True
    return False

def start():
    if status():
        return True
    
    Popen(
        f"sudo {tool_path} --no-virt -o {interface_to_internet} --ap {interface_to_create_ap} my_ap -p 12345678".split(' '),
        stdout=DEVNULL, stderr=DEVNULL
    )
    time.sleep(3)
    
    if status():
        return True
    return False

def get_ap_pid():
    ps = run(f"{tool_path} --list-running".split(' '), capture_output=True, encoding='utf-8')
    for line in ps.stdout.split('\n'):
        if interface_to_create_ap in line:
            pid = line.split(' ')[0]
            print(f"pid={pid}")
            return int(pid)
    return -1

def stop():
    if not status():
        return True
    
    pid = get_ap_pid()
    if pid == -1:
        return True
    ps = Popen(f"sudo {tool_path} --stop {pid}".split(' '), stdout=DEVNULL, stderr=DEVNULL)
    time.sleep(3)

    if not status():
        return True
    return False
```

Log in python command line:
```
lin@ubuntu:~/tool$ sudo python
Python 3.8.10 (default, Sep 28 2021, 16:10:42) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from manage_ap import *
>>> status()
False
>>> start()
True
>>> status()
True
>>> stop()
pid=50825
True
>>> Terminated
lin@ubuntu:~/tool$ 
```

And `ps axfj`:
(1) when I enter python command line and not run lnxrouter
```
   3731    4707    4707    4707 ?             -1 Ssl   1000   0:10  \_ /usr/libexec/gnome-terminal-server
   4707   12369   12369   12369 pts/1      51150 Ss    1000   0:00  |   \_ bash
  12369   51150   51150   12369 pts/1      51150 S+       0   0:00  |   |   \_ sudo python
  51150   51151   51150   12369 pts/1      51150 S+       0   0:00  |   |       \_ python
```
(2) after I start wifi AP in python command line
```
   3731    4707    4707    4707 ?             -1 Ssl   1000   0:10  \_ /usr/libexec/gnome-terminal-server
   4707   12369   12369   12369 pts/1      51150 Ss    1000   0:00  |   \_ bash
  12369   51150   51150   12369 pts/1      51150 S+       0   0:00  |   |   \_ sudo python
  51150   51151   51150   12369 pts/1      51150 S+       0   0:00  |   |       \_ python
  51151   51179   51150   12369 pts/1      51150 S+       0   0:00  |   |           \_ sudo ./lnxrouter --no-virt -o ens33 --ap wlx1cbfce3736c6 my_a
  51179   51180   51150   12369 pts/1      51150 S+       0   0:00  |   |               \_ /bin/bash ./lnxrouter --no-virt -o ens33 --ap wlx1cbfce37
  51180   51283   51150   12369 pts/1      51150 S+       0   0:00  |   |                   \_ cat /dev/shm/lnxrouter_tmp/lnxrouter.wlx1cbfce3736c6.
  51180   51310   51150   12369 pts/1      51150 S+       0   0:00  |   |                   \_ /bin/bash ./lnxrouter --no-virt -o ens33 --ap wlx1cbf
  51310   51367   51150   12369 pts/1      51150 S+       0   0:00  |   |                   |   \_ sleep 2
  51180   51315   51150   12369 pts/1      51150 S+       0   0:00  |   |                   \_ hostapd -P /dev/shm/lnxrouter_tmp/lnxrouter.wlx1cbfce
  51180   51316   51150   12369 pts/1      51150 S+       0   0:00  |   |                   \_ /bin/bash ./lnxrouter --no-virt -o ens33 --ap wlx1cbf
  51316   51363   51150   12369 pts/1      51150 S+       0   0:00  |   |                   |   \_ sleep 10
  51180   51355   51150   12369 pts/1      51150 S+       0   0:00  |   |                   \_ bash -c while :; do sleep 8000 ; done 
  51355   51356   51150   12369 pts/1      51150 S+       0   0:00  |   |                       \_ sleep 8000
```
(3) after I stop wifi AP in python command line
```
   3731    4707    4707    4707 ?             -1 Ssl   1000   0:10  \_ /usr/libexec/gnome-terminal-server
   4707   12369   12369   12369 pts/1      12369 Ss+   1000   0:00  |   \_ bash
```

garywill, 2021-11-10:
It is a bug. We shouldn't use PGID to kill.

Delete these lines, then the bug should gone
https://github.com/garywill/linux-router/blob/15a2e0ce53928c658d15b651d03cf4a88a460d8c/lnxrouter#L1176-L1178
(and need to observe if some children processes not terminated after exit)

closed 2024-12-30 by garywill