garywill/linux-router #29: Hotspot successfully created - Cannot connect from any device - Obtaining IP Address

francescopapu, 2021-11-28:
Hello. I've been trying to setup a working hotspot for quite some time now.
Internet is provided to the laptop via USB-A ethernet adapter, laptop has Realtek wifi interface that supports AP mode and other modes. Running on KDE Neon (with Plasma) latest version. WIFI Hotspot worked flawlessly on Windows 10.

I would like to post logs of what happens when i try to connect from Android phone / other devices, but don't know how to log (complete newbie). I tried to run the lnxrouter command in various ways, but can't manage to get any device connected, Android phone shows that it is stuck on "obtaining ip address".

garywill, 2021-11-30:
try:
    1) 'firewall-cmd --zone=trusted --add-interface=<SUBN_IFACE>'
    2) disable firewalld

francescopapu, 2021-11-30:
It seems like i don't have 'firewalld' installed
> Command 'firewall-cmd' not found, but can be installed with:
> sudo apt install firewalld

What should i do?

francescopapu, 2021-11-30:
> try: 1) 'firewall-cmd --zone=trusted --add-interface=<SUBN_IFACE>' 2) disable firewalld

It seems like I don't have firewalld installed, since it says:
```
Command 'firewall-cmd' not found, but can be installed with:
sudo apt install firewalld
```

closed 2024-12-30 by garywill