garywill/linux-router #31: script reconfiguring local interface and cutting off local ssh

varoudis, 2022-01-12:
Hi,

I cant find a way to run the script without killing my own link to the system.

On the 'router': 
eth0 is on 192.168.166.100 (Im connecting to it from 192.168.166.110 via ssh)
wlan0 is the link to the internet (no firewalls, no VPNs etc.. simple wifi).

I tried both:
`sudo lnxrouter -i eth0 -o wlan0`
and:
`sudo lnxrouter -i eth0 -o wlan0 -g 192.168.166.100`

Both killed my link to the 'router' from eth0.
Am I missing something?
Thanks

garywill, 2022-01-13:
I guess this is the network you're trying to make:
```
Internet ---[ wlan0 - router - eth0 ]----- computers
```
Am I getting it right?

The current script `-i eth0` will reset eth0. 
To solve this, we need to implement an option to not reset in future version.

BTW, currently can you reconnect to router via ssh after running the command and it resetting?

**Before the fix is implemented, there's  workaround.**
**Try comment out these:**
https://github.com/garywill/linux-router/blob/f6995bc30c5b85ac07aefe37b35004a307e3f62b/lnxrouter#L2008-L2032
https://github.com/garywill/linux-router/blob/f6995bc30c5b85ac07aefe37b35004a307e3f62b/lnxrouter#L1162
https://github.com/garywill/linux-router/blob/f6995bc30c5b85ac07aefe37b35004a307e3f62b/lnxrouter#L1158
https://github.com/garywill/linux-router/blob/f6995bc30c5b85ac07aefe37b35004a307e3f62b/lnxrouter#L1169

Then run
```
sudo lnxrouter -i eth0  -g 192.168.166.100
```

(**Notice** this is a dirty workaround for your current use case only, not a true fix) 
I haven't tested.

varoudis, 2022-01-13:
hi and thanks for the quick response!

This is my actual setup.
`Internet ---[ wlan0 - router (PiZero2) - usb0 ]----- just one computer - ethernet over USB`
I have the feeling that the reset if killing the usb0 (gadget mode) and is not coming back at all...

Ill check you temp fix.
Best
Tasos

varoudis, 2022-01-19:
@garywill 

This is what I got:

`sudo ./lnxrouter -i usb0 -g 192.168.166.100                                                                                          1 ⨯
linux-router 0.6.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 1075
Target interface is usb0 (3a:ea:66:54:bc:a2)
RTNETLINK answers: File exists
Error occured

ERROR: Failed setting usb0 IPv4 address

Doing cleanup..
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done`

and:

`sudo ./lnxrouter -i usb0 -o wlan0 -g 192.168.166.100
[sudo] password for kali:
linux-router 0.6.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 991
Target interface is usb0 (3a:ea:66:54:bc:a2)
RTNETLINK answers: File exists
Error occured

ERROR: Failed setting usb0 IPv4 address

Doing cleanup..
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done`

With your edits above.
Thanks

garywill, 2022-01-20:
Sorry for inconvinence. Try branch https://github.com/garywill/linux-router/tree/workaround-31