garywill/linux-router #33: Network traffic is forwarded to tor, but onion network services don't work!

JhonasHenrique, 2022-03-10:
Network traffic is forwarded to tor perfectly using the following command: ./lnxrouter --ap interface SSID -p PASSWORD --tp 9040 --dns 9053 but onion network services, that is, .onion sites do not open no way. Could you fix this? I use your script specifically for this purpose and I think this functionality is extremely important. 

I get the following error when accessing onion network sites, my tor gives me this as output whenever I send a request to .onion services: error: [warn] Onion address [scrubbed] requested from a port with .onion disabled 

Sincerely: Jonas.

garywill, 2022-03-12:
1.  https://tor.stackexchange.com/questions/15536/what-does-the-logged-tor-warning-onion-address-foo-requested-from-a-port-with 
2. `lnxrouter --log-dns`  see what tor's DNS answer to .onion
3. If successfully make tor use virtual IP for .onion, according to the answer:
    > "virtual address" is in the CIDR IP range 127.192.0.0/10 by default

    We need to make some change in our script https://github.com/garywill/linux-router/blob/f6995bc30c5b85ac07aefe37b35004a307e3f62b/lnxrouter#L1030 .
    Or change tor's IP range for .onion to a non-LAN range (if to avoid some engineering jobs )
    Guys any better ideas also welcomed

JhonasHenrique, 2022-03-14:
I tried to make the changes proposed by you! But I was not successful! My torrc file is well configured, follow its configuration:

VirtualAddrNetwork 10.192.0.0/10
AutomapHostsSuffixes .onion, .exit
AutomapHostsOnResolve 1
Transport 0.0.0.0:9040
DNSPort 0.0.0.0:9053

As you can see the configuration above is already configured to automatically resolve the onion domains ".onion, .exit" but without success! I was here doing some tests, and when I do a manual configuration only on the device that the tor is installed on I was successful in accessing .onion services but it was forwarding via port 9050 that it uses socks. Obviously I tried to forward the tp to 9050 but it was to be expected that it wouldn't work. Being able to forward traffic via tor and having a hostspot that supports multiple devices is very cool, but not being able to use onion services makes this option a little unlimited. But there are other scripts similar to this one like: https://github.com/htrgouvea/nipe anon-surf: https://github.com/Und3rf10w/kali-anonsurf kali-torify: https://github.com/brainfucksec /kalitorify they make a transparent proxy and can enjoy the services of the onion network, but unlike its alternative, the transparent proxy using these tools is limited to only one user as it does not serve as a bridge to other devices.

JhonasHenrique, 2022-03-14:
Espero que com sua ajuda e da comunidade possamos resolver isso! Bem, é muito interessante.

JhonasHenrique, 2022-03-14:
I hope that with your help and the community we can solve this! Well, it's very interesting.

garywill, 2022-03-15:
> VirtualAddrNetwork 10.192.0.0/10

That's still LAN range. 

Try `VirtualAddrNetwork 20.192.0.0/10`. 
See if that makes `.onion` work.

JhonasHenrique, 2022-03-15:
Have you ever managed to access onion network services with your script? I will try again, and I will not rest until I resolve this. If you can help me with my research I would appreciate it. Well I'll try to apply what you just said in your last answer. xD

JhonasHenrique, 2022-03-15:
The problem was solved when I added 20.192.0.0/10 to my torrc file I changed it from VirtualAddrNetwork 10.192.0.0/22 to VirtualAddrNetwork wonderful! Now I can use onion network services. Now I understand, I was setting a class a ip with cidr that ended up in my network when the resolution would have to be done in wan. Right? Can you clarify the solution further? And if possible, indicate me some material about the reserved blocks of the tor guides and etc... I'm very happy!

garywill, 2022-03-15:
Actually the problem was our script wouldn't redirect LAN traffic to proxy
https://github.com/garywill/linux-router/blob/f6995bc30c5b85ac07aefe37b35004a307e3f62b/lnxrouter#L1022-L1035
WAN is not necessary for tor.

 I don't know much about tor, this is the first time user asking about .onion webs.

garywill, 2022-03-15:
Using 20.x.x.x will have potential conflict with some public websites.
According to https://en.wikipedia.org/wiki/IPv4#Special-use_addresses
```
198.18.0.0/15    Used for benchmark testing of inter-network communications between two separate subnets.[15]

240.0.0.0/4  Reserved for future use.[18] (Former Class E network.) 
```
can be considered as map range.

Any one have better ideas? I did not ever use ip mapping

JhonasHenrique, 2022-03-16:
rom documentation:

    When Tor needs to assign a virtual (unused) address because of a MAPADDRESS command from the controller or the AutomapHostsOnResolve feature, Tor picks an unassigned address from this range. (Defaults: 127.192.0.0/10 and [FE80::]/10 respectively.) When providing proxy server service to a network of computers using a tool like dns-proxy-tor, change the IPv4 network to "10.192.0.0/10" or "172.16.0.0/12" and change the IPv6 network to "[FC00::]/7". The default VirtualAddrNetwork address ranges on a properly configured machine will route to the loopback or link-local interface. The maximum number of bits for the network prefix is set to 104 for IPv6 and 16 for IPv4. However, a wider network - smaller prefix length is preferable since it reduces the chances for an attacker to guess the used IP. For local use, no change to the default VirtualAddrNetwork setting is needed.

JhonasHenrique, 2022-03-16:
Como os dispositivos que estão conectados ao roteador linux não possuem o tor instalado, ou seja, não rodando sozinho, pois estão participando de um proxy transparente, acabei de colocar o seguinte no meu torrc ao invés do VirtualAdresNetwork 10.192.0.0/10 I coloque VirtualAdresNetwork (Ip local da máquina rodando linux-router/tor) então se precisar rotear algum pacote, ele será roteado para a própria máquina, e os serviços onion estão funcionando perfeitamente. A princípio tentei configurar o gateway gerado pelo roteador linux, mas a configuração para o arquivo torrc VirtualAddrNetwork só suporta cidr /16 e o ip do meu gateway era classe c ip mas ultrapassou o número de bits possíveis. Acho que a solução para todo esse conflito é baixar o gateway para classe a com cidr /10 ou classe b cidr /16 para este pequeno trecho que tirei acima da documentação você terá uma limpeza mental e depois resolverá o problema. xD

closed 2024-12-30 by garywill