garywill/linux-router #36: Your adapter can not transmit to channel 36, frequency band 5GHz.

RealTheSunCat, 2022-04-06:
Hi! I found this tool while looking for a solution to create an access point with my AX200 PCIe card.
I'm looking to connect my Quest 2 VR headset (which supports WiFi 6) directly to my computer with as little latency as possible, and no access to the external internet. My understanding is that a 5GHz hotspot should provide this, so I ran the following command:
```console
$ sudo lnxrouter -n --ap wlan0 MyAccessPoint -p passphrase --freq-band 5
linux-router 0.6.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 19700
Target interface is wlan0 (92:b6:95:3b:19:81)
Use random LAN IPv4 address 192.168.53.1
Creating a virtual WiFi interface... 
x0wlan0 created
Error occured

ERROR: Your adapter can not transmit to channel 36, frequency band 5GHz.

Doing cleanup.. 
Undoing iptables changes ..
bash: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.rGInnm/undo_iptables.sh: No such file or directory
Exiting: This is the only running instance
Killing PGID 19699 ...
Cleaning up done
```

I had the same issue trying to use `linux-wifi-hotspot`, so I think it's an issue with `create_ap`, but it seems it is now unmaintained.
Is there anything I can try? The 2.4GHz hotspot gives me 58Mbps, which is not what the AX200 card I bought advertises.

I've attached the output of a few other commands:
```console
$ lspci | grep AX200
03:00.0 Network controller: Intel Corporation Wi-Fi 6 AX200 (rev 1a)

$ iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 44:af:28:36:70:72
                type managed
                txpower 0.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```
[iw_list.txt](https://github.com/garywill/linux-router/files/8426823/iw_list.txt)

HuggieBo, 2023-04-23:
Hey,
I have the exact same issue.
I've got it working by using nm-connection-editor to create a new 5ghz hotspot and then launching it with
> nmcli connection up hotspot

TheSunCat, 2023-05-02:
Thank you! I have since bought a dedicated router, so I do not need to create a hotspot anymore, but at least this solves it for people who come across this issue.

rajivharlalka, 2023-09-02:
@HuggieBo Hi , I am facing similar issues, can you detail about ho you setup 5ghz hotspot using nm-connection-editor