garywill/linux-router #40: Script fails without ipv6

Jabbado, 2022-07-13:
I tried running this on a Puppy Linux distro and it fails while backing up settings with
`ERROR: Failed backing up interface ipv6 bits`

There's no /proc/sys/net/ipv6 directory on this system. I don't see an option to disable ipv6 in the script, only to enable it.
Any way of getting this to work?

garywill, 2022-07-14:
> I tried running this on a Puppy Linux distro

hmmm... it seems a shrinked system, not a complete Linux.

This script is designed for modern completed Linux, so I'm afraid we won't do some change to fit your requirement. However I think you can try deleting some IPv6 code (as below) in script:
https://github.com/garywill/linux-router/blob/9e1d985623e92a507c6eba93e0bfddb25e3e6f01/lnxrouter#L1121
https://github.com/garywill/linux-router/blob/9e1d985623e92a507c6eba93e0bfddb25e3e6f01/lnxrouter#L2037
https://github.com/garywill/linux-router/blob/9e1d985623e92a507c6eba93e0bfddb25e3e6f01/lnxrouter#L1132

Jabbado, 2022-07-15:
Thank you yes I thought commenting out those lines would get it working too. But after doing so it turned out Puppy Linux is just too streamlined and missing too many dependencies. Testing on Lubuntu however seems to be working well.

closed 2022-07-15 by Jabbado