garywill/linux-router #44: command failed: Invalid argument (-22)

Cabbache, 2022-09-24:
When trying to serve a hotspot on the same interface that has internet, the script is failing. ```sudo ./lnxrouter --ap <wireless interface with internet> hotspot -p password```.

```
linux-router 0.6.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 216391
Target interface is <interface with internet> (a6:1a:58:36:f3:c9)   PCI: 
Use random LAN IPv4 address 192.168.99.1
<interface with internet> already in channel 6 (2437 MHz)
Channel fallback to 6
Creating a virtual WiFi interface... 
kernel reports: Attribute failed policy validation
command failed: Invalid argument (-22)
Error occured

ERROR: Failed creating virtual WiFi interface. Maybe your WiFi adapter does not fully support virtual interfaces. Try again with '--no-virt'
```
If I add --no-virt it works but no internet is served on the hotspot. I had no trouble doing this with the original create_ap script.

The problem is coming from this [line](https://github.com/garywill/linux-router/blob/master/lnxrouter#L1648).

I even tried manually running the command iw dev ...etc but it returns the same error:
```
kernel reports: Attribute failed policy validation
command failed: Invalid argument (-22)
```

Cabbache, 2022-09-24:
I've confirmed that it works when I make the name of the virtual interface shorter instead of 2 characters added to the original interface name.

garywill, 2022-09-25:
Hi, thank you for reporting the issue and sending PR!

I think it is better to defaultly name the new interface name after the original interface name. 
People may create more than 1 virtual interface based on different wifi adapters simutaneously.

So adding an option to rename the new interface (e.g. `--virt-name`) would be better. 
Also, the script should warn user if the original interface has a long name.

Cabbache, 2022-10-30:
Currently the `alloc_new_vface_name` function seems to increment `$i` until `$VIRT_PREFIX$i` is not the name of an existing interface. If we change it to `v_iface_name="$VIRT_NAME"`, then should the script exit with an error if the interface already exists?

garywill, 2022-10-31:
Yes. User should receive an error if the interface name already exist.

PR #45 (2022-09-27): renamed virtual interface
If you have a ${WIFI_IFACE} that has a long name, it will cause [this](https://github.com/garywill/linux-router/blob/master/lnxrouter#L1648) line to fail. I have renamed the virtual interface to something shorter. Fixes: #44
closed 2023-01-07 by garywill