garywill/linux-router #47: addr_gen_mode no such file or directory

RalphCodesTheInternet, 2022-10-04:
Currently when starting the access point, I get the following output.

linux-router 0.6.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 4714
Target interface is wlan0 (30:d8:c4:52:d8:91) 
cp: cannot stat '/proc/sys/net/ipv6/conf/wlan0/addr_gen_mode': No such file or directory
Error occured

ERROR: Failed backing up interface ipv6 bits

Doing cleanup.. 
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done

Before this was working on Ubuntu 22, but now I have downgraded to 20.04 and receiving this issue. Any tips?

garywill, 2022-10-04:
Try delete

https://github.com/garywill/linux-router/blob/a414ac035b3bc8edbdac594fc49f5eed185a2f70/lnxrouter#L1086
https://github.com/garywill/linux-router/blob/a414ac035b3bc8edbdac594fc49f5eed185a2f70/lnxrouter#L1099

RalphCodesTheInternet, 2022-10-04:
I can confirm `/proc/sys/net/ipv6/conf/wlan0/addr_gen_mode` does not exist.

I have tried Ubuntu desktop as well as server, with the same issue.
I could try using Ubuntu 18 and see if it works better?

If commenting out those two lines, other errors appear, such as

cp: target '/proc/sys/net/ipv6/conf/wlan0/use_tempaddr' is not a directory
./lnxrouter: line 1087: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.2L0RT0/sys_6_conf_iface/: Is a directory
Error occured

ERROR: Failed backing up interface ipv6 bits

Strange that this happens on Ubuntu 20. Ive used this script on Ubuntu 22, Armbian and Raspbian operating systems without any problems.

RalphCodesTheInternet, 2022-10-04:
What seemed to work was using another kernel version of Ubuntu. I was using 4.9. When upgrading to Focal with kernel 5.13, it works perfectly. Can confirm that it works on Ubuntu Jammy with kernel 5.16 as well. So i guess the problem was related to the older version of the Linux kernel. 

So for anyone having the same issue, I guess upgrading to anything 5+ should do the trick.

Thanks for the help in any case @garywill it is much appreciated.

closed 2022-10-04 by RalphCodesTheInternet