garywill/linux-router #51: How to create 5G AP ?

CaiJingLong, 2022-11-27:
My network card is `3168NGW`.
I know that it supports 5G.

I use `lnxrouter --ap wlp2s0 HelloWIFI -p 12345678` to create AP. 
But the wifi only support 2.4G.

garywill, 2022-11-27:
Hi, there

Sorry for currently our script doesn't have 5G support. I don't have a 5G adapter to do tests. 

You can try some other create_ap forks (e.g. https://github.com/lakinduakash/linux-wifi-hotspot) to see if they have added that support.

Actually also I'm expecting someone send a PR to add 5G support to linux-router.

CaiJingLong, 2022-11-27:
Thanks for your help, I try to use the [linux-wifi-hotspot](https://github.com/lakinduakash/linux-wifi-hotspot).
I can't even open the AP use linux-wifi-hotspot. 

Host anyone to open PR to support 5G.

zqlee-ronghui, 2022-12-08:
I have a MT7612U adaptor, and I can get hotspot with `sudo lnxrouter -ap wlan0 apppp --ieee80211n  --freq-band 5` under Ubuntu 20.04. 

The question here is I can't connect to the AP if I don't set a static IP for the client, I guess the reason maybe similar to https://github.com/garywill/linux-router/issues/19

garywill, 2022-12-10:
`--freq-band 5` is from create_ap. I myself didn't tried it ever. I may work. maybe~

---------------------------------------

@zqlee-ronghui 

> The question here is I can't connect to the AP if I don't set a static IP for the client, I guess the reason maybe similar to #19

If it's because of firewalld, do this
https://github.com/garywill/linux-router/blob/a414ac035b3bc8edbdac594fc49f5eed185a2f70/lnxrouter#L862

keogami, 2024-02-27:
I have just tried `--freq-band 5` and it seems to be working fine

closed 2024-12-30 by garywill