garywill/linux-router #53: Can't change the network ip address

justin-mottier, 2023-01-05:
I'm trying to set a custom network address, but unfortunately it never changes.

I'm running the access point using this command:
`sudo ./lnxrouter --ap ap@wlan0 MyAccessPoint -p MyPassPhrase -g 10.10.10.10`

When I look at my ip address, I have this:
```
4: ap@wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether e4:5f:01:5c:51:af brd ff:ff:ff:ff:ff:ff
    inet 192.168.4.1/24 brd 192.168.4.255 scope global ap@wlan0
       valid_lft forever preferred_lft forever
    inet 10.10.10.10/24 brd 10.10.10.255 scope global ap@wlan0
       valid_lft forever preferred_lft forever
```

The first one never changes, regardless of what I give in -g or even if i set -g.

I can ping 10.10.10.10 from the host itself, but not from a connected client.

Concerning the clients, they got an a dress in 192.168.4.0 too:
```
3: wlp59s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether a8:7e:ea:30:ad:09 brd ff:ff:ff:ff:ff:ff
    inet 192.168.4.49/24 brd 192.168.4.255 scope global dynamic noprefixroute wlp59s0
       valid_lft 3597sec preferred_lft 3597sec
    inet6 fe80::30ce:702d:cbdf:a5cb/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```

garywill, 2023-01-07:
What OS distro and version?
I guess there's another DHCP or some services running on your host .(Try find them and stop them first.)
Post the logs.

justin-mottier, 2023-01-09:
I'm running Ubuntu 18 on a raspberry pi 4

After searching for DHCP services I didn't find anything (I searched for lease file in /var/lib/dhcp but I don't have a dhcp directory, and I didn't find any services that looked like DHCP services)

Here are the lnxrouter logs:

```
linux-router 0.6.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 21847
Target interface is ap@wlan0 (e4:5f:01:5c:51:af) 

haveged_watchdog PID: 21911

Starting hostapd
WARN: Low entropy detected. We recommend you to install `haveged'
hostapd PID: 21916
Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.ap@wlan0.conf.rZvHp6/hostapd.conf
ap@wlan0: Could not connect to kernel driver
Using interface ap@wlan0 with hwaddr e4:5f:01:5c:51:af and ssid "toast"
ap@wlan0: interface state UNINITIALIZED->ENABLED
ap@wlan0: AP-ENABLED 

iptables v1.8.4 (legacy)
NOTICE: Not showing all operations done to iptables rules

iptables: NAT 
MASQUERADE  all opt -- in * out !ap@wlan0  10.10.10.0/24 !-> 10.10.10.0/24   /* lrt21847ap@wlan0 */
ACCEPT  all opt -- in ap@wlan0 out *  10.10.10.0/24  -> 0.0.0.0/0   /* lrt21847ap@wlan0 */
ACCEPT  all opt -- in * out ap@wlan0  0.0.0.0/0  -> 10.10.10.0/24   /* lrt21847ap@wlan0 */
Loaded kernel module nf_nat_pptp

iptables: allow DNS
ACCEPT  tcp opt -- in ap@wlan0 out *  10.10.10.0/24  -> 10.10.10.10   tcp dpt:53 /* lrt21847ap@wlan0 */
ACCEPT  udp opt -- in ap@wlan0 out *  10.10.10.0/24  -> 10.10.10.10   udp dpt:53 /* lrt21847ap@wlan0 */

iptables: allow dhcp
ACCEPT  udp opt -- in ap@wlan0 out *  0.0.0.0/0  -> 0.0.0.0/0   udp dpt:67 /* lrt21847ap@wlan0 */

Starting dnsmasq
Jan  9 09:21:24 dnsmasq[21957]: started, version 2.80 cachesize 150
Jan  9 09:21:24 dnsmasq[21957]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth nettlehash DNSSEC loop-detect inotify dumpfile
Jan  9 09:21:24 dnsmasq-dhcp[21957]: DHCP, IP range 10.10.10.10 -- 10.10.10.250, lease time 1h
Jan  9 09:21:24 dnsmasq-dhcp[21957]: DHCP, sockets bound exclusively to interface ap@wlan0
Jan  9 09:21:24 dnsmasq[21957]: reading /etc/resolv.conf
Jan  9 09:21:24 dnsmasq[21957]: using nameserver 127.0.0.53#53
Jan  9 09:21:24 dnsmasq[21957]: cleared cache
dnsmasq PID: 21957

== Setting up completed, now linux-router should be working ==
```

garywill, 2023-01-09:
What's the full command you ran? 
Did you use `--no-virt`? Your log shows it did not create new virtual interface.

If that's true, `ap@wlan0` must be created and was being handled by another service.

justin-mottier, 2023-01-09:
This is the full command I use:
`sudo ./lnxrouter --ap ap@wlan0 MyAccessPoint -p MyPassPhrase -g 10.10.10.10`

You're right, I've created a virtual interface myself using this command:
`sudo /sbin/iw dev wlan0 interface add ap@wlan0 type __ap`

I use this virtual device in order to still have wlan0 as a classic wifi client.

Moreover, this is maybe an other issue, but if I'm trying to run the hotspot with on wlan0, here is what I get (with logs):
`sudo ./lnxrouter --ap wlan0 MyAccessPoint -p MyPassPhrase -g 10.10.10.10 --log-dns --hostapd-debug 2`
```
WARN: brmfmac driver doesn't work properly with virtual interfaces and
      it can cause kernel panic. For this reason we disallow virtual
      interfaces for your adapter.
      For more info: https://github.com/oblique/create_ap/issues/203
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
linux-router 0.6.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 132973
Target interface is wlan0 (e4:5f:01:5c:51:af) 

haveged_watchdog PID: 133040

Starting hostapd
WARN: Low entropy detected. We recommend you to install `haveged'
hostapd PID: 133046
random: getrandom() support available
Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.rF5LKe/hostapd.conf
ctrl_interface_group=0
nl80211: Using driver-based roaming
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Try mode change after setting interface down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Failed to set interface 3 to mode 3: -16 (Device or resource busy)
nl80211: Delaying mode set while interface going down
[... 65 lines cut ...]
Terminated
Killed haveged_watchdog.pid 133040 lnxrouter
Terminated
Killed keep_running.pid 133119 bash
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done
```
Then my bash prompt appears, but some logs are displayed after that:
```
Error occured

ERROR: dnsmasq failed

./lnxrouter: line 1232: kill: (132973) - No such process

Doing cleanup.. 
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done
```

It looks like two instances are run.

Regarding the errors in the logs (`Failed to set interface 3 to mode 3: -16 (Device or resource busy)`), I didn't find any process which could keep the interface busy, even if I stop the netplan service.

garywill, 2023-01-09:
1. Have you tried a virtual interface name without `@`? just letters only
1.  `ss -tlunp`  list listening ports. Let's see what services  probably running?
1. If you run with a manually created ap interface while `wlan0` as client , try some different `--freq-band <GHz>` and `-c <channel>` combinations. Ap and wlan0 having same or different channels.... See will that make some difference
2. Since it is warning about imcompatible driver , which I'm afraid I can't help with. Maybe we can wait for some people's help

justin-mottier, 2023-01-09:
Wow, using `apwlan` as an an interface name seems to work, I have my ip well set to 10.10.10.XXX, and so are the devices connected to it. Do you know what can cause this ?

Just in case, here is the output of `ss -tlunp`:

```
Netid     State      Recv-Q     Send-Q                             Local Address:Port            Peer Address:Port     Process                                                                                                                
udp       UNCONN     0          0                                        0.0.0.0:46647                0.0.0.0:*         users:(("tf2_web_republi",pid=1112,fd=8))                                                                             
udp       UNCONN     0          0                                        0.0.0.0:44906                0.0.0.0:*         users:(("move_group",pid=1126,fd=8))                                                                                  
udp       UNCONN     0          0                                  127.0.0.53%lo:53                   0.0.0.0:*                                                                                                                               
udp       UNCONN     0          0                            192.168.0.170%wlan0:68                   0.0.0.0:*                                                                                                                               
udp       UNCONN     0          0                              192.168.1.98%eth0:68                   0.0.0.0:*                                                                                                                               
udp       UNCONN     0          0                                        0.0.0.0:47241                0.0.0.0:*         users:(("niryo_robot_har",pid=1124,fd=8))                                                                             
udp       UNCONN     0          0                                        0.0.0.0:57510                0.0.0.0:*         users:(("rosout",pid=1017,fd=8))                                                                                      
udp       UNCONN     0          0                                        0.0.0.0:51626                0.0.0.0:*         users:(("robot_state_pub",pid=1122,fd=8))                                                                             
udp       UNCONN     0          0                                        0.0.0.0:50264                0.0.0.0:*         users:(("python3",pid=1127,fd=17))                                                                                    
udp       UNCONN     0          0               [fe80::e65f:1ff:fe5c:51af]%wlan0:546                     [::]:*                                                                                                                               
tcp       LISTEN     0          5                                        0.0.0.0:46183                0.0.0.0:*         users:(("python3",pid=1135,fd=7))                                                                                     
tcp       LISTEN     0          128                                      0.0.0.0:41031                0.0.0.0:*         users:(("roslaunch",pid=851,fd=4))                                                                                    
tcp       LISTEN     0          128                                      0.0.0.0:5000                 0.0.0.0:*         users:(("python3",pid=847,fd=13))                                                                                     
tcp       LISTEN     0          128                                      0.0.0.0:39273                0.0.0.0:*         users:(("python3",pid=1127,fd=4))                                                                                     
tcp       LISTEN     0          128                                      0.0.0.0:37609                0.0.0.0:*         users:(("python3",pid=1142,fd=4))                                                                                     
tcp       LISTEN     0          5                                        0.0.0.0:33289                0.0.0.0:*         users:(("python3",pid=1153,fd=7))                                                                                     
tcp       LISTEN     0          128                                      0.0.0.0:44201                0.0.0.0:*         users:(("python3",pid=1160,fd=4))                                                                                     
tcp       LISTEN     0          5                                        0.0.0.0:36905                0.0.0.0:*         users:(("robot_state_pub",pid=1122,fd=9))                                                                             
tcp       LISTEN     0          128                                      0.0.0.0:42955                0.0.0.0:*         users:(("python3",pid=1153,fd=4))                                                                                     
tcp       LISTEN     0          128                                      0.0.0.0:39499                0.0.0.0:*         users:(("python3",pid=847,fd=4))                                                                                      
tcp       LISTEN     0          128                                      0.0.0.0:39693                0.0.0.0:*         users:(("python3",pid=1113,fd=7))                                                                                     
[... cut ...]
tcp       LISTEN     0          5                                        0.0.0.0:34277                0.0.0.0:*         users:(("python3",pid=1046,fd=7))                                                                                     
tcp       LISTEN     0          5                                        0.0.0.0:43717                0.0.0.0:*                                                                                                                               
tcp       LISTEN     0          128                                         [::]:22                      [::]:*                                                   
```

closed 2024-12-30 by garywill