garywill/linux-router #58: hostapd error

dca00, 2023-08-08:
Hello,

Do these errors mean that my system is missing something? What should I do to arrive at a working hotspot?

Thank you for your help!

```
$ sudo ./lnxrouter --ap wlan0 ********* -p *****************
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
which: no hostapd in (/sbin:/bin:/usr/sbin:/usr/bin)
linux-router 0.6.7 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 2633
Target interface is wlan0 (20:10:7a:7a:4d:2b)   PCI: 0000:03:00.0 Network controller [0280]: Ralink corp. RT5390 Wireless 802.11n 1T/1R PCIe [1814:5390]
Use random LAN IPv4 address 192.168.38.1

haveged_watchdog PID: 2725

Starting hostapd
hostapd PID: 2730
/bin/stdbuf: failed to run command 'hostapd': No such file or directory

iptables v1.6.2
NOTICE: Not showing all operations done to iptables rules

iptables: NAT 
MASQUERADE  all opt -- in * out !wlan0  192.168.38.0/24 !-> 192.168.38.0/24   /* lrt2633wlan0 */
ACCEPT  all opt -- in wlan0 out *  192.168.38.0/24  -> 0.0.0.0/0   /* lrt2633wlan0 */
ACCEPT  all opt -- in * out wlan0  0.0.0.0/0  -> 192.168.38.0/24   /* lrt2633wlan0 */
Loaded kernel module nf_nat_pptp

iptables: allow DNS
ACCEPT  tcp opt -- in wlan0 out *  192.168.38.0/24  -> 192.168.38.1   tcp dpt:53 /* lrt2633wlan0 */
ACCEPT  udp opt -- in wlan0 out *  192.168.38.0/24  -> 192.168.38.1   udp dpt:53 /* lrt2633wlan0 */

iptables: allow dhcp
ACCEPT  udp opt -- in wlan0 out *  0.0.0.0/0  -> 0.0.0.0/0   udp dpt:67 /* lrt2633wlan0 */

Starting dnsmasq
Aug  8 15:30:36 dnsmasq[2782]: started, version 2.76 cachesize 150
Aug  8 15:30:36 dnsmasq[2782]: compile time options: IPv6 GNU-getopt DBus no-i18n IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC loop-detect inotify
Aug  8 15:30:36 dnsmasq-dhcp[2782]: DHCP, IP range 192.168.38.10 -- 192.168.38.250, lease time 1h
Aug  8 15:30:36 dnsmasq-dhcp[2782]: DHCP, sockets bound exclusively to interface wlan0
Aug  8 15:30:36 dnsmasq[2782]: reading /etc/resolv.conf
Aug  8 15:30:36 dnsmasq[2782]: using nameserver 10.10.10.1#53
Aug  8 15:30:36 dnsmasq[2782]: cleared cache
dnsmasq PID: 2782

== Setting up completed, now linux-router should be working ==
Error occured

ERROR: hostapd failed

Error occured

Doing cleanup.. 
Aug  8 15:30:44 dnsmasq[2782]: exiting on receipt of SIGTERM
Killed dnsmasq.pid 2782 dnsmasq
./lnxrouter: line 771:  2811 Terminated              sleep 2
Killed haveged_watchdog.pid 2725 lnxrouter
Failed to kill hostapd.pid 2730 , it may have exited
bash: line 1:  2806 Terminated              sleep 8000
Killed keep_running.pid 2805 bash
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done
[use0@localhost Compile]$ Error occured

ERROR: dnsmasq failed

./lnxrouter: line 1231: kill: (2633) - No such process

Doing cleanup.. 
Undoing iptables changes ..
Exiting: This is the only running instance
Cleaning up done

```

garywill, 2023-08-10:
> failed to run command 'hostapd': No such file or directory

The log already said, couldn't find `hostapd`. Install it

dca00, 2023-08-10:
Very cool, thanks! Everything works now. Kudos to you for the great tool!

closed 2023-12-08 by garywill