garywill/linux-router #65: Unable to connect to hotspot

TobiasDax, 2023-11-02:
Tried running this on a raspberry pi 4 and at first it seemed to work but i am unable to connect to the Hotspot with my Android Phone or windows PC

i tried to run it with the default settings given in the readme but nothing would connect, then i saw the --hs20 flg and ran that.
At first it looked like the same no connection error, but then the whole tool crashed

Here is the complete output

```
~/linux-router $ sudo ./lnxrouter --ap wlan0 MyAccessPoint -p MyPassPhrase --hs20
WARN: brmfmac driver doesn't work properly with virtual interfaces and
      it can cause kernel panic. For this reason we disallow virtual
      interfaces for your adapter.
      For more info: https://github.com/oblique/create_ap/issues/203
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
linux-router 0.7.1b (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 18716
Target interface is wlan0 (dc:a6:32:53:c0:26)
Use random LAN IPv4 address 192.168.63.1
Channel not specified, use default
Set wlan0 unmanaged by NetworkManager

haveged_watchdog PID: 18825

Starting hostapd
hostapd PID: 18831
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED

iptables v1.8.9 (nf_tables)
NOTICE: Not showing all operations done to iptables rules

iptables: NAT
MASQUERADE  all opt -- in * out !wlan0  192.168.63.0/24 !-> 192.168.63.0/24   /* lrt18716wlan0 */
ACCEPT  all opt -- in wlan0 out *  192.168.63.0/24  -> 0.0.0.0/0   /* lrt18716wlan0 */
ACCEPT  all opt -- in * out wlan0  0.0.0.0/0  -> 192.168.63.0/24   /* lrt18716wlan0 */
Loaded kernel module nf_nat_pptp

iptables: allow DNS
ACCEPT  tcp opt -- in wlan0 out *  192.168.63.0/24  -> 192.168.63.1   tcp dpt:53 /* lrt18716wlan0 */
ACCEPT  udp opt -- in wlan0 out *  192.168.63.0/24  -> 192.168.63.1   udp dpt:53 /* lrt18716wlan0 */

iptables: allow dhcp
ACCEPT  udp opt -- in wlan0 out *  0.0.0.0/0  -> 0.0.0.0/0   udp dpt:67 /* lrt18716wlan0 */

Starting dnsmasq
Nov  2 22:09:24 dnsmasq[18867]: started, version 2.89 cachesize 150
Nov  2 22:09:24 dnsmasq[18867]: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC loop-detect inotify dumpfile
Nov  2 22:09:24 dnsmasq-dhcp[18867]: DHCP, IP range 192.168.63.10 -- 192.168.63.250, lease time 1h
Nov  2 22:09:24 dnsmasq-dhcp[18867]: DHCP, sockets bound exclusively to interface wlan0
Nov  2 22:09:24 dnsmasq[18867]: reading /etc/resolv.conf
Nov  2 22:09:24 dnsmasq[18867]: using nameserver 127.0.0.1#53
Nov  2 22:09:24 dnsmasq[18867]: cleared cache
dnsmasq PID: 18867

== Setting up completed, now linux-router should be working ==
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 6e:0d:e3:c2:48:9b
wlan0: STA 6e:0d:e3:c2:48:9b RADIUS: starting accounting session 87379B80C8056720
wlan0: STA 6e:0d:e3:c2:48:9b WPA: pairwise key handshake completed (RSN)
wlan0: EAPOL-4WAY-HS-COMPLETED 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPDISCOVER(wlan0) 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPOFFER(wlan0) 192.168.63.109 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPDISCOVER(wlan0) 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPOFFER(wlan0) 192.168.63.109 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPDISCOVER(wlan0) 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPOFFER(wlan0) 192.168.63.109 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPREQUEST(wlan0) 192.168.63.109 6e:0d:e3:c2:48:9b
Nov  2 22:10:01 dnsmasq-dhcp[18867]: DHCPACK(wlan0) 192.168.63.109 6e:0d:e3:c2:48:9b Pixel-6a
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 6e:0d:e3:c2:48:9b
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.11: disassociated
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.11: associated
wlan0: AP-STA-POSSIBLE-PSK-MISMATCH 6e:0d:e3:c2:48:9b
wlan0: STA 6e:0d:e3:c2:48:9b WPA: received EAPOL-Key msg 2/4 in invalid state (10) - dropped
wlan0: STA 6e:0d:e3:c2:48:9b WPA: received EAPOL-Key msg 2/4 in invalid state (10) - dropped
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.11: disassociated
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 6e:0d:e3:c2:48:9b
wlan0: STA 6e:0d:e3:c2:48:9b RADIUS: starting accounting session 03D4083AA987A829
wlan0: STA 6e:0d:e3:c2:48:9b WPA: pairwise key handshake completed (RSN)
wlan0: EAPOL-4WAY-HS-COMPLETED 6e:0d:e3:c2:48:9b
Nov  2 22:10:12 dnsmasq-dhcp[18867]: DHCPREQUEST(wlan0) 192.168.63.109 6e:0d:e3:c2:48:9b
Nov  2 22:10:12 dnsmasq-dhcp[18867]: DHCPACK(wlan0) 192.168.63.109 6e:0d:e3:c2:48:9b Pixel-6a
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 6e:0d:e3:c2:48:9b
Error occured

ERROR: hostapd failed

./lnxrouter: line 1: 18831 Segmentation fault      $STDBUF_PATH hostapd $HOSTAPD_DEBUG_ARGS -P "$CONFDIR/hostapd.pid" "$CONFDIR/hostapd.conf"
Error occured

Doing cleanup..
Nov  2 22:10:22 dnsmasq[18867]: exiting on receipt of SIGTERM
Killed dnsmasq.pid 18867 dnsmasq
Terminated
Killed haveged_watchdog.pid 18825 lnxrouter
Failed to kill hostapd.pid 18831 , it may have exited
Terminated
Killed keep_running.pid 18896 bash
Undoing iptables changes ..
Error occured

ERROR: dnsmasq failed

```

garywill, 2023-11-03:
Usually use default options. `--hs20` is not needed.
Your phone doesn't work with `-w 1` apparently. `-w 2` seems ok. Just use it without `-w`, use default.

For now I can't tell which is going wrong, try add `--hostapd-debug 1` ( or `2`). Let's see if the debug log can tell us something

TobiasDax, 2023-11-03:
Yeah, i started with the default options and just tried other stuff when those did not work.

tried again with `--hostapd-debug-1`

```
~/linux-router $ sudo ./lnxrouter --ap wlan0 MyAccessPoint -p MyPassPhrase --hostapd-debug 1
WARN: brmfmac driver doesn't work properly with virtual interfaces and
      it can cause kernel panic. For this reason we disallow virtual
      interfaces for your adapter.
      For more info: https://github.com/oblique/create_ap/issues/203
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
linux-router 0.7.1b (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 79906
Target interface is wlan0 (dc:a6:32:53:c0:26)
Use random LAN IPv4 address 192.168.112.1
Channel not specified, use default
Set wlan0 unmanaged by NetworkManager

haveged_watchdog PID: 80014

Starting hostapd
hostapd PID: 80020
random: getrandom() support available
Configuration file: /dev/shm/lnxrouter_tmp/lnxrouter.wlan0.conf.Kt34RC/hostapd.conf
ctrl_interface_group=0
nl80211: Using driver-based roaming
nl80211: TDLS supported
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Using driver-based off-channel TX
nl80211: Supported vendor command: vendor_id=0x1018 subcmd=1
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: Enable multi-channel concurrent (driver advertised support)
nl80211: use P2P_DEVICE support
nl80211: key_mgmt=0xd0f enc=0x10f auth=0x7 flags=0x2000000380cfacc rrm_flags=0x0 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=10
nl80211: interface wlan0 in phy phy0
nl80211: Set mode ifindex 3 iftype 3 (AP)
nl80211: Setup AP(wlan0) - device_ap_sme=1 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x559f6efb60 (device SME)
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=04 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=0501 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=0503 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=0504 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=06 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=08 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=0a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=11 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=12 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x559f6efb60 match=7f multicast=0
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x559f6efb60 match= multicast=0
nl80211: Enable Probe Request reporting nl_preq=0x559f6f4830
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x559f6f4830 match= multicast=0
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 3 (ifidx_reason -1)
nl80211: if_indices[16]: 3(-1)
phy: phy0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
nl80211: Regulatory information - country=DE (DFS-ETSI)
nl80211: 2400-2483 @ 40 MHz 20 mBm
nl80211: 5150-5250 @ 80 MHz 23 mBm (no outdoor)
nl80211: 5250-5350 @ 80 MHz 20 mBm (no outdoor) (DFS)
nl80211: 5470-5725 @ 160 MHz 26 mBm (DFS)
nl80211: 5725-5875 @ 80 MHz 13 mBm
nl80211: 5945-6425 @ 160 MHz 23 mBm (no outdoor)
[... 2257 lines cut ...]
nl80211: Set STA flags - ifname=wlan0 addr=6e:0d:e3:c2:48:9b total_flags=0x0 flags_or=0x0 flags_and=0xfffffff1 authorized=0
wlan0: STA 6e:0d:e3:c2:48:9b WPA: event 2 notification
wpa_driver_nl80211_set_key: ifindex=3 (wlan0) alg=0 addr=0x5576627ad0 key_idx=0 set_tx=1 seq_len=0 key_len=0 key_flag=0x20
nl80211: DEL_KEY
   addr=6e:0d:e3:c2:48:9b
   pairwise key
nl80211: set_key failed; err=-22 Invalid argument
RSN: PTK removal from the driver failed
WPA: 6e:0d:e3:c2:48:9b WPA_PTK entering state DISCONNECTED
WPA: 6e:0d:e3:c2:48:9b WPA_PTK entering state INITIALIZE
wpa_driver_nl80211_set_key: ifindex=3 (wlan0) alg=0 addr=0x5576627ad0 key_idx=0 set_tx=1 seq_len=0 key_len=0 key_flag=0x20
nl80211: DEL_KEY
   addr=6e:0d:e3:c2:48:9b
   pairwise key
nl80211: set_key failed; err=-22 Invalid argument
RSN: PTK removal from the driver failed
nl80211: Set STA flags - ifname=wlan0 addr=6e:0d:e3:c2:48:9b total_flags=0x0 flags_or=0x0 flags_and=0xfffffffe authorized=0
wlan0: STA 6e:0d:e3:c2:48:9b IEEE 802.1X: unauthorizing port
nl80211: Set STA flags - ifname=wlan0 addr=6e:0d:e3:c2:48:9b total_flags=0x0 flags_or=0x0 flags_and=0xfffffff1 authorized=0
nl80211: sta_remove -> DEL_STATION wlan0 6e:0d:e3:c2:48:9b --> 0 (Success)
ap_free_sta: cancel ap_handle_timer for 6e:0d:e3:c2:48:9b
```

the only thing really installed on my raspberry is casaOS, when i find a spare SD card i will try again with just a fresh install, to rule out some problems there...
Thanks for your time

garywill, 2023-11-04:
```
WPA: EAPOL-Key MIC using HMAC-SHA1
Searching a PSK for 6e:0d:e3:c2:48:9b prev_psk=0x559f6f59e0
wlan0: STA 6e:0d:e3:c2:48:9b WPA: invalid MIC in msg 2/4 of 4-Way Handshake
```

```
wlan0: STA 6e:0d:e3:c2:48:9b WPA: received EAPOL-Key 2/4 Pairwise with unexpected replay counter
```
seems your devices can't finish the 2/4 handshake.

I don't know the reason behind that ...

My log after 2/4 handshake is 
```
WPA: EAPOL-Key MIC using HMAC-SHA1
WPA: xxxxxxxx WPA_PTK entering state PTKCALCNEGOTIATING2
WPA: xxxxxxxx WPA_PTK entering state PTKINITNEGOTIATING
x0wlan0: STA xxxxxxxx WPA: sending 3/4 msg of 4-Way Handshake
```

garywill, 2023-11-04:
Have you tried shorter ssid and password?

TobiasDax, 2023-11-04:
I tried a lot of different SSIDs and passwords, no help there…

I also tried setting up a hotspot over nmcli, nmtui and over RasAP. All did not work. 
I tried it after a fresh install of raspberry pi os lite and after some use, no help.

then yesterday in an act of frustration I installed the full raspberry OS and tried it over the Desktop UI … and it works. Just to be sure I also tried it over nmcli and nmtui and both also work…
I also ran `rpi-update` on the full os, so I can’t really be sure what fixed it, the non-lite os version or the rpi-update.
But one of those did the trick. When I have time next week I will try an other lite install with rpi-update to check and a full install without just to be sure.

but yeah, that’s how I got it to finally work with the built in raspberry tools. After that I did not try your Skript again since I did not want to break anything :) 
But I will also test that next week to give you some feedback.

thanks for your time

closed 2024-12-30 by garywill