garywill/linux-router #66: nftables support

tiix95, 2023-11-06:
Hello,

Thank you for this code as it really helps. I can see you supports iptables but not nftables (it may work in some environments with iptables-nft). Would you be interested in a PR for a support like this ?

I managed to make it work for my use case using native nft rules, and I'll share this if you think it might be helpful.

garywill, 2023-11-08:
Hi, @tiix95 

You can share your work here of cource.

Although I believe someday we will have to switch to nftables, when most popular distros deprecate legacy iptables (years later, I think) , for now I think linux-router script would keep using legacy, for compatibility.

Feel free to talk if you have different opinion.

Switching will require a lot of work. I guess there will be breaking changes and not 100% current features are ensured to be available. I'm not full-time sysadmin nor CS/IT worker. That will cost me much time to learn new rules and syntax. 
When the day come, I'll appreciate if anyone help with that. (as unfortunately I'm the only maintainer of this project -_-~)

tiix95, 2023-11-08:
You can find attach the patch for the 0.7.1b version. As far I can see, all is working well using iptables *or* nftables, this patch adds the support for nftables and aims to not break anything using iptables.

 I still am currently testing to check if my patch didn't break anything,  feel free to check the patch on your own if you want :) I did not test redsocks for now.

Regarding the support, I actually am full-time sysadmin and I use iptables / nftables all day long. If you need any support for this, or other linux/network stuff, i'd be glad to help.

I also patched some lines for the code to be shellcheck compliant :)

[patch_lnxrouter.txt](https://github.com/garywill/linux-router/files/13298767/patch_lnxrouter.txt)

cafinux, 2024-02-24:
I'm learning to use nftables and patch_lnxrouter.txt is one cool script. Do you have it or some script like it you run against machines which do some functions as router? Pulling chunks out of the patch will be really handy as there's a serious lack of examples of in production scripts for nftables on the net. By the way I'm finding nftables infinitely easier to use that iptables as their syntax is so much cleaner and more natural to use. Sorry for jumping in on this forum for my own interests but I think this would enhance lnxrouter alot because of the readability of nftables when debugging etc. If there was a version with this set of patches applied I would love to test it.