garywill/linux-router #67: iw version 6.7 prints frequency differently

kphanipavan, 2024-01-07:
[image]

Current frequency taken from `iw` now prints differently now, as shown above.

Changes related to that is shown in the diff: https://fossies.org/diffs/iw/5.19_vs_6.7/info.c-diff.html

The number after the dot is the frequency offset. The awk script requires changes to handle this. Below is the error thrown currently.

[image]

~Pvn

kphanipavan, 2024-01-07:
Before:
[image]

After, using `cut`:
[image]

Works on my end.

garywill, 2024-01-09:
Thank you for reporting. We should take reaction to that change.

I'm still using a distro that using iw 4.9. So are you or anyone willing to make a PR?

petebueh, 2024-01-10:
I copied the changes from this commit from linux-wifi-hotspot:
https://github.com/lakinduakash/linux-wifi-hotspot/commit/2dc1649add81854b79903dd16d59fcbbfa2fd8d3

This works for my AP

closed 2024-01-19 by garywill

garywill, 2024-01-19:
https://github.com/garywill/linux-router/commit/32f168ec8c9a69e022ae97d3bfd113fe088626d6 
I made `can_transmit_to_channel()` ignores freq value and just focus on channel number

```bash
grep -E " [0-9]+(\.[0-9]+){0,1} MHz \[${CHANNEL_NUM}\]"
```

PR #68 (2024-01-19): FIX #67: Remove trailing .0 from iw reported frequency
This should fix #67 
The sed command removes a dot followed by any number (greater than zero) of zeros followed by the end of the line. This way frequencies now reported with a trailing .0 will also work.
Just in case it will also remove .00 and similar, but not ".", ".5" and similar because they may actually have a different meaning and thus should not be rounded away as suggested in #67 with cut but throw an error.
The more complete approach would be to use software like bc to do actual non-integer comparison but as long as there are no  [... cut]