garywill/linux-router #72: Wifi 6E 6GHZ Support

lilkidsuave, 2024-03-18:
There's no 6ghz option.

jernejmarcic, 2024-07-24:
Check if you are even "allowed" to have a 6Ghz hotspot with `sudo iw phy`. For some ~~bullshit~~ reasons I for one cannot

kphanipavan, 2026-02-20:
As of now, Feb 2026, hotspot on 6GHz can be turned on in EU and GB. `iw reg set GB` or `iw reg set FR` should work.

kphanipavan, 2026-05-11:
[image]

Work in progress, seems to be buggy. 6GHz wifi hotspot launched my lnxrouter patch. 

[image]

I am testing on borrowed hardware and my phone's 6ghz is not enabled yet, so need more testers' reports than before.