garywill/linux-router #75: doesn't work when VPN is connected

shwewo, 2024-03-27:
for some reason `generate_random_ip4()` inside of this script stales when you are connected to the VPN. i don't know why, maybe because `is_ip4_lan_range_available $random_ip4 && break` loops and can't find available ip address.

solution is to specify gateway manually:

```bash
sudo lnxrouter -g 10.0.0.1 --ap wlp1s0 <ssid> -p <password>
```

shwewo, 2024-03-28:
also, i can't really specify network interface for wifi AP when i'm connected to VPN. specifying network interface manually when connected results in broken internet.  

i can see packets from the clients, they are routing to the host, but not outside of the host. 

Works fine though if i don't specify it. Routes fine to VPN subnet.

i'm manually setting output interface like that: `lnxrouter -o enp3s0f3u1u1 --no-dns --dhcp-dns 1.1.1.1 --ap wlp1s0 ...`

garywill, 2024-03-29:
Hi, thank you for feedback.

For the 2 issues you described, please try both latest commit (0.7.6b) and 0.7.3. 
I recently did many changes so we need to check if it is those changes that make things go different.

> for some reason generate_random_ip4() inside of this script stales when you are connected to the VPN. i don't know why, maybe because is_ip4_lan_range_available $random_ip4 && break loops and can't find available ip address.

Could you provide the output of following commands, when connected to VPN
1. `ip a`
2. `ip r`

shwewo, 2024-03-29:
Hi!
i tried 0.7.6b, it doesn't work too.

i'm launching it like that:

```
sudo lnxrouter -g 10.0.0.1 --country RU -o enp3s0f3u1u1 --no-dns  --dhcp-dns 1.1.1.1 --ap wlp1s0 laptop -p password --freq-band 5 --wifi4 --wifi5 --no-virt --qr
```

`ip a` output:

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute 
       valid_lft forever preferred_lft forever
2: wlp1s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether d6:3f:0f:df:03:7a brd ff:ff:ff:ff:ff:ff permaddr 8c:c8:4b:d6:0c:d5
5: virbr0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 52:54:00:b4:01:f9 brd ff:ff:ff:ff:ff:ff
    inet 192.168.122.1/24 brd 192.168.122.255 scope global virbr0
       valid_lft forever preferred_lft forever
25: enp3s0f3u1u1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:e0:4c:68:61:73 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.76/24 brd 192.168.1.255 scope global dynamic noprefixroute enp3s0f3u1u1
       valid_lft 86035sec preferred_lft 86035sec
    inet6 fe80::91c2:c1f8:2912:a0ad/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
27: warp0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet 172.16.0.2/32 scope global noprefixroute warp0
       valid_lft forever preferred_lft forever
```

(warp0 works, i have ping/dns, i also tried your script with tailscale exit node, it doesn't work when specifying output interface).

`ip r` output:

```
default via 192.168.1.1 dev enp3s0f3u1u1 proto dhcp src 192.168.1.76 metric 100 
10.0.0.0/24 dev wlp1s0 proto kernel scope link src 10.0.0.1 
192.168.1.0/24 dev enp3s0f3u1u1 proto kernel scope link src 192.168.1.76 metric 100 
192.168.122.0/24 dev virbr0 proto kernel scope link src 192.168.122.1 linkdown 
```

`ip -6 r` output:

```
fe80::/64 dev enp3s0f3u1u1 proto kernel metric 1024 pref medium
```

Also, i have tried doing `iptables -F`  when running your script. Sadly, it doesn't help too

garywill, 2024-03-29:
If `enp3s0f3u1u1` is your physical interface which you get Internet from, and `warp0` is the virtual interface created after you connect to VPN, `-o enp3s0f3u1u1` is wrong, `-o warp0` is what you want.

shwewo, 2024-03-29:
> If `enp3s0f3u1u1` is your physical interface which you get Internet from, and `warp0` is the virtual interface created after you connect to VPN, `-o enp3s0f3u1u1` is wrong, `-o warp0` is what you want.

i don't quite understand the logic. I want to share internet from interface enp3s0f3u1u1, i don't want my clients connected to VPN. I want my system connected to a VPN, and tether wifi without vpn using my default ISP (thats why i want to specify  enp3s0f3u1u1 manually, e.g. firejail understands this logic firejail --net=enp3s0f3u1u1 chromium and it shows my default ISP IP)

garywill, 2024-03-29:
> I want my system connected to a VPN, and tether wifi without vpn using my default ISP 

Sorry, I got wrong. 

Does `-o enp3s0f3u1u1` work for your wifi clients when VPN is not connected?
And then `-o enp3s0f3u1u1` stops working after VPN connected.
If so, which route are other apps on system using defaultly? If you run Chromium directly, what IP doesn it get?

Firejail uses different machanism than iptables (which our script uses).

shwewo, 2024-03-29:
> Does -o enp3s0f3u1u1 work for your wifi clients when VPN is not connected?

yes, it does

> And then -o enp3s0f3u1u1 stops working after VPN connected.

yes, it stops working after VPN is connected

> If so, which route are other apps on system using defaultly? If you run Chromium directly, what IP doesn it get?

they use VPN route by default (e.g. if i run chromium directly it shows warp0 ip, not my ISP ip address)

> Firejail uses different machanism than iptables (which our script uses).

i understand, iirc it uses cgroups, but thats out of scope. i'm just saying how i imagine things in my head so you can understand this analogy.

shwewo, 2024-03-29:
> I guess VPN program wants itself to be default route and is preventing traffics going through enp3s0f3u1u1

I use plain wireguard with nmcli if this could help. More complex VPNs tinker with routing/firewall rules very heavily so i explicitly used more simpler one that works out-of-box for this report.

garywill, 2024-03-29:
>  firejail --net=enp3s0f3u1u1 chromium and it shows my default ISP IP)

You could try `sudo  firejail --net=enp3s0f3u1u1` then run lnxrouter in it. See what result

shwewo, 2024-03-29:
> You could try `sudo firejail --net=enp3s0f3u1u1` then run lnxrouter in it. See what result

I understand that it can't work (not possible because it isolates the network interfaces, so wlp1s0 shouldn't be available inside of firejail sandbox). 

I actually tried moving wireless interface to a different network namespace too, but it turns out the driver has to support this.

garywill, 2024-03-29:
Maybe creating a sandbox to let VPN run in there and share to outside can solve your need.

BTW, could you do `sudo bash -x lnxrouter .....` to see why `generate_random_ip4` doesn't work?

solsticedhiver, 2024-07-19:
same problem here when I simply to try to set up a "router" between 2 ethernet interfaces.

Without mullvad VPN running, this is working fine.
With it running, lnxrouter does not complete set up

with either using `lnxrouter -o eth0` or with `lnxrouter -o mullvad` (which is the VPN interface)