garywill/linux-router #77: Can't use 'iw' to operate interfce 'wlan0'

GovenorKratos, 2024-05-17:
Whenver I try to use the script, I get this issue:
`WARN: Can't use 'iw' to operate interfce 'wlan0', trying 'iwconfig' (not as good as 'iw') ... (Did you spell the interface name right?)`
`ERROR: Can't use 'iwconfig' to operate interfce 'wlan0'` (iwconfig is not in Fedora's repos)

I'm on Fedora 40. The command I used was:
`sudo ./linux-router.sh --ap wlan0 MyAccessPoint -p MyPassPhrase
`

dca00, 2024-06-09:
You are missing hostapd package: https://github.com/garywill/linux-router/issues/58

osamwelian3, 2025-04-12:
@dca00 Really. `hostapd` when the error clearly says `iw` and `iwconfig`. I am getting the same error and am sure beyond reasonable doubt that I have hostapd installed

GovenorKratos, 2025-04-12:
It’s been a very long time since I created this issue and I’ve since sold the previous laptop and switched to Mac. From what I remember, I managed to get it working by turning it into a shell script and running it with sudo.

What command did you use? Is your network interface actually wlan0? To double check, you can use the ifconfig command in your shell.

osamwelian3, 2025-04-12:
I got it to work

> It’s been a very long time since I created this issue and I’ve since sold the previous laptop and switched to Mac. From what I remember, I managed to get it working by turning it into a shell script [...]

I realized my network interface was actually not wlan0 to start with but even that did not fix it, it led to another error that said the device was busy. I then found out I could turn off the network interface with 

`nmcli radio wifi off`

then: sudo ./linux-router.sh --ap wlan0 MyAccessPoint -p MyPassPhrase 

this would successfully create the ap/wireless hotspot

then bring wifi back on with:

`nmcli radio wifi off`

Connect to the wifi network and you are able to share your internet. Its a long process but it works.