garywill/linux-router #84: lspci: -s: Invalid bus number

solsticedhiver, 2024-10-30:
When just running ` lnxrouter --ap wlan0 Test -p TestTestTest -c 6 -n`,I got

```linux-router 0.7.6 (https://github.com/garywill/linux-router)
Released under LGPL, with no warranty. Use on your own risk.

PID: 217668
Target interface is wlan0 (a2:b5:33:83:00:d0) 
lspci: -s: Invalid bus number
PCI: 1-11.2:1.0
System-already-loaded driver: ath9k_htc
```

and no ap up

garywill, 2024-11-16:
Sorry for late reply.

That could be caused by that the device path is not like the script expects.
https://github.com/garywill/linux-router/blob/e7ecf8707d04cfcf5d9bb8363b9e3f717f58889f/lnxrouter#L693-L704

Please use `sudo bash -x ./lnxrouter ......` to run and get the debug log.

What's your system and version?

solsticedhiver, 2024-11-16:
the system is archlinux (up to date) with a tp-link usb wifi card (722N)

[lnxrouter.log](https://github.com/user-attachments/files/17784547/lnxrouter.log)

garywill, 2024-11-16:
> Warning: Extension comment revision 0 not supported, missing kernel module?

That may be the reason you couldn't get ap up.
Try this:
```bash
env NETFILTER_XT_MATCH_COMMENT=0 ./lnxrouter --ap ........
```

--------------------------

You're using usb card. `lspci` only does with pci devices. That could be the reason that it showed Invalid bus number. That's not important.

solsticedhiver, 2024-11-16:
yes, it works with your last command.

I have compiled my own old kernel too, 6.1.117-1-lts61, and not module have been compiled