garywill/linux-router #87: WPA3 support (SAE)

redfast00, 2024-12-19:
Thank you for this project, it has come in useful quite often while developing our own open-source Wi-Fi stack on the ESP32 (see https://github.com/esp32-open-mac).

I saw that there was no WPA3 support yet; would it be possible to add that? I'd be interested in both the handshake (SAE) and in the GCMP encryption mode.

garywill, 2024-12-20:
Yes, WPA3 is in the plan. But I'm not sure when 🤷‍♂️ (it's been in the plan for a long time)
Theorically simply adding some options into `hostapd.conf` will do that. 

https://github.com/garywill/linux-router/blob/fe2348a725246f3ed83e0f31a4330bfce5e2eb38/lnxrouter#L1877-L1887

Recently I don't have spare time to do or test. You can try. I'm also expecting PR

kphanipavan, 2026-02-20:
@redfast00 can you test WPA3 with my patch #97 ?